dresden-elektronik/deconz-rest-plugin #7369: Sonoff SNZB-06P

lhering, 2023-11-21:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Sonoff SNZB-06P Human Presence Sensor

### Manufacturer

Sonoff

### Model identifier

SNZB-06P

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest
[image]

[image]

[image]

lhering, 2023-11-24:
I was playing with the DDF files and got a working version:
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "SNZB-06P",
  "vendor": "Sonoff",
  "product": "SNZB-06P",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/snzb-06p.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/resetpresence",
          "default": false
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "refresh.interval": 4,
          "read": {
            "fn": "None"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0406",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0406",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x18",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}`

BabaIsYou, 2023-11-29:
Does it also work whitout all this stuff for `state/presence` ?

```
"refresh.interval": 4,
"read": {
"fn": "None"
},
"parse": {
"at": "0x0000",
"cl": "0x0406",
"ep": 1,
"eval": "Item.val = Attr.val",
"fn": "zcl"
},
"default": false
```

lhering, 2023-11-29:
> Does it also work whitout all this stuff for `state/presence` ?

Yes, still working without and the binding stuff is also not needed.
This is the current DDF file:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "SNZB-06P",
  "vendor": "Sonoff",
  "product": "SNZB-06P",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/snzb-06p.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/dark",
          "refresh.interval": 30,
          "read": {
            "at": "0x2001",
            "cl": "0xfc11",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x2001",
            "cl": "0xfc11",
            "ep": 1,
            "eval": "if (Attr.val === 0) { Item.val = true } else { Item.val = false }",
            "fn": "zcl"
          },
          "default": false
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "default": false
        }
      ]
    }
  ]
}
``` 
I've asked the Sonoff Support about the 2 unknown cluster FC11 and FC57:
_
FC11 attr:0x2001 type:uint8 value:0 dim 1 bright.
FC57 Z2M does not need this._

So I added this to the general.xml:
```
    <!-- SONOFF -->
    <cluster id="0xfc11" name="Sonoff specific" mfcode="0x1286">
      <description>Cluster for light sensor</description>
      <server>
        <attribute id="0x2001" name="Light" type="u8" access="r" required="m" mfcode="0x1286">
          <description></description>
        </attribute>
      </server>
      <client>
      </client>
    </cluster>
``` 

I think the name "Light" should be changed but I'm not shure of a better name.

BabaIsYou, 2023-11-30:
Can you delete these lines one more time and just this binding part ?

```
     "bindings": [
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
	  {
		"bind": "unicast",
		"src.ep": 1,
		"cl": "0x0406",
"		report": [
			{
			"at": "0x0000",
			"dt": "0x18",
			"min": 1,
			"max": 300,
			"change": "0x00000001"
			}
				]
	  }
	  ]
```

lhering, 2023-11-30:
With that config the state still doesnt change. When I click the read Attribute button in the IAS Zone Cluster then the state shows correctly.

Current DDF:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "SNZB-06P",
  "vendor": "Sonoff",
  "product": "SNZB-06P",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/snzb-06p.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/dark",
          "refresh.interval": 30,
          "read": {
            "at": "0x2001",
            "cl": "0xfc11",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x2001",
            "cl": "0xfc11",
            "ep": 1,
            "eval": "if (Attr.val === 0) { Item.val = true } else { Item.val = false }",
            "fn": "zcl"
          },
          "default": false
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "default": false
        }
      ]
    }
  ],
     "bindings": [
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
	  {
		"bind": "unicast",
		"src.ep": 1,
		"cl": "0x0406",
                "report": [
			{
			"at": "0x0000",
			"dt": "0x18",
			"min": 1,
			"max": 300,
			"change": "0x00000001"
			}
				]
	  }
	  ]
}
```

Idaho947, 2023-12-05:
Hello, 
Any news about this integration. I wonder if I won't buy a few of them

lhering, 2023-12-05:
> Hello, Any news about this integration. I wonder if I won't buy a few of them

This DDF is currently working fine for me:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "SNZB-06P",
  "vendor": "Sonoff",
  "product": "SNZB-06P",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/snzb-06p.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/dark",
          "refresh.interval": 30,
          "read": {
            "at": "0x2001",
            "cl": "0xfc11",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x2001",
            "cl": "0xfc11",
            "ep": 1,
            "eval": "if (Attr.val === 0) { Item.val = true } else { Item.val = false }",
            "fn": "zcl"
          },
          "default": false
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "refresh.interval": 4,
          "read": {
            "fn": "None"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0406",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          },
          "default": false        }
      ]
    }
  ],
     "bindings": [
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
	  {
		"bind": "unicast",
		"src.ep": 1,
		"cl": "0x0406",
                "report": [
			{
			"at": "0x0000",
			"dt": "0x18",
			"min": 1,
			"max": 300,
			"change": "0x00000001"
			}
				]
	  }
	  ]
}
```

hackepeterOli, 2023-12-14:
Hi, i have a new SNZB-06P, but with the above ddf it does not work. Any Ideas?
[image]
[image]
[image]
[image]

Mimiix, 2023-12-14:
> Hi, i have a new SNZB-06P, but with the above ddf it does not shown in Phoscon App. Any Ideas? ![Sonoff SNZB-06P](https://private-user-images.githubusercontent.com/44376749/290650390-d6c37959-549a-471b-a550-3f1f4e06551d.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1ODgwODcsIm5iZiI6MTcwMjU4Nzc4NywicGF0aCI6Ii80NDM3Njc0OS8yOTA2NTAzOTAtZDZjMzc5NTktNTQ5YS00NzFiLWE1NTAtM2YxZjRlMDY1NTFkLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE0VDIxMDMwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMxMjdlZDM1NDZiYmFiMTgzNjllNThhYTUwYTFiNDM5ZTI5ODRmOGVlNGVkYWI1OThkOWQ0MDM3MmFlM2Y2OTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.2ICQBCgVknsoD3M_kr6LXt6KXQS-tqQX-4k_fbkQ3gg) ![Sonoff SNZB-06P_1](https://private-user-images.githubusercontent.com/44376749/290650393-931c9d6c-f126-416e-aabd-65318ad7189c.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1ODgwODcsIm5iZiI6MTcwMjU4Nzc4NywicGF0aCI6Ii80NDM3Njc0OS8yOTA2NTAzOTMtOTMxYzlkNmMtZjEyNi00MTZlLWFhYmQtNjUzMThhZDcxODljLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE0VDIxMDMwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY1ODRiYTkyZGZkMzhmYmQzM2JhMTFmMmFkOWY5YWY5NGE3OTQ5NjZjZjVlZTY3ZjkyMTI0MzE0NDVlZDQ4MzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qDlAouRa9fZutWCE8SiugXE0BSvpSOWBh80r4tU0nWU) ![Sonoff SNZB-06P_2](https://private-user-images.githubusercontent.com/44376749/290650396-262b013f-2bea-4f9d-a828-296093baf40a.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1ODgwODcsIm5iZiI6MTcwMjU4Nzc4NywicGF0aCI6Ii80NDM3Njc0OS8yOTA2NTAzOTYtMjYyYjAxM2YtMmJlYS00ZjlkLWE4MjgtMjk2MDkzYmFmNDBhLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE0VDIxMDMwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk3NmY5ZDJiMDlkMjhhODJhZTQzMTRlY2UwN2NmYmIwNzY5ZGE0OGMwMTUxY2UxMzcyYjgwNmM0NGVlNWIwYjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vgpDoOBXsYf3I-t1RhPXBDpQjEBJ3BvyqR08YWOa23c) ![Sonoff SNZB-06P_3](https://private-user-images.githubusercontent.com/44376749/290650399-0b33ea1d-ec74-4590-9957-05f7995d09ab.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1ODgwODcsIm5iZiI6MTcwMjU4Nzc4NywicGF0aCI6Ii80NDM3Njc0OS8yOTA2NTAzOTktMGIzM2VhMWQtZWM3NC00NTkwLTk5NTctMDVmNzk5NWQwOWFiLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE0VDIxMDMwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQwOTRiNWMxMTFkY2RhM2ZhOGEzNjlmOTk3ZmVmN2Q4YWY0OTljOTI1ZTJlNzExYzhkYTE0YTY2ZTgxMDk5MTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.yMhYGWiGNrhHQX3ExHZb8YBY5z79JzyZxFSDHW7Fo6Y)

Did you enable bronze ddfs to be loaded ?

If you don't know how, change the bronze to gold in the DDF and reload.

hackepeterOli, 2023-12-14:
> Did you enable bronze ddfs to be loaded ?
>
> If you don't know how, change the bronze to gold in the DDF and reload.

Thanks, now it is shown in Phoscon App - but the Motion or Presence is not shown:

[image]

Mimiix, 2023-12-14:
> > > Hi, i have a new SNZB-06P, but with the above ddf it does not shown in Phoscon App. Any Ideas? ![Sonoff SNZB-06P](https://private-user-images.githubusercontent.com/44376749/290650390-d6c37959-549a-471b-a550-3f1f4e06551d.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1ODgwODcsIm5iZiI6MTcwMjU4Nzc4NywicGF0aCI6Ii80NDM3Njc0OS8yOTA2NTAzOTAtZDZjMzc5NTktNTQ5YS00NzFiLWE1NTAtM2YxZjRlMDY1NTFkLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE0VDIxMDMwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMxMjdlZDM1NDZiYmFiMTgzNjllNThhYTUwYTFiNDM5ZTI5ODRmOGVlNGVkYWI1OThkOWQ0MDM3MmFlM2Y2OTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.2ICQBCgVknsoD3M_kr6LXt6KXQS-tqQX-4k_fbkQ3gg) ![Sonoff SNZB-06P_1](https://private-user-images.githubusercontent.com/44376749/290650393-931c9d6c-f126-416e-aabd-65318ad7189c.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1ODgwODcsIm5iZiI6MTcwMjU4Nzc4NywicGF0aCI6Ii80NDM3Njc0OS8yOTA2NTAzOTMtOTMxYzlkNmMtZjEyNi00MTZlLWFhYmQtNjUzMThhZDcxODljLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE0VDIxMDMwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY1ODRiYTkyZGZkMzhmYmQzM2JhMTFmMmFkOWY5YWY5NGE3OTQ5NjZjZjVlZTY3ZjkyMTI0MzE0NDVlZDQ4MzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qDlAouRa9fZutWCE8SiugXE0BSvpSOWBh80r4tU0nWU) ![Sonoff SNZB-06P_2](https://private-user-images.githubusercontent.com/44376749/290650396-262b013f-2bea-4f9d-a828-296093baf40a.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1ODgwODcsIm5iZiI6MTcwMjU4Nzc4NywicGF0aCI6Ii80NDM3Njc0OS8yOTA2NTAzOTYtMjYyYjAxM2YtMmJlYS00ZjlkLWE4MjgtMjk2MDkzYmFmNDBhLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE0VDIxMDMwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk3NmY5ZDJiMDlkMjhhODJhZTQzMTRlY2UwN2NmYmIwNzY5ZGE0OGMwMTUxY2UxMzcyYjgwNmM0NGVlNWIwYjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vgpDoOBXsYf3I-t1RhPXBDpQjEBJ3BvyqR08YWOa23c) ![Sonoff SNZB-06P_3](https://private-user-images.githubusercontent.com/44376749/290650399-0b33ea1d-ec74-4590-9957-05f7995d09ab.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1ODgwODcsIm5iZiI6MTcwMjU4Nzc4NywicGF0aCI6Ii80NDM3Njc0OS8yOTA2NTAzOTktMGIzM2VhMWQtZWM3NC00NTkwLTk5NTctMDVmNzk5NWQwOWFiLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE0VDIxMDMwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQwOTRiNWMxMTFkY2RhM2ZhOGEzNjlmOTk3ZmVmN2Q4YWY0OTljOTI1ZTJlNzExYzhkYTE0YTY2ZTgxMDk5MTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.yMhYGWiGNrhHQX3ExHZb8YBY5z79JzyZxFSDHW7Fo6Y)
> > 
> > 
> > Did you enable bronze ddfs to be loaded ?
> > If you don't know how, change the bronze to gold in the DDF and reload.
> 
> Thanks, now it is shown in Phoscon App - but the Motion or Presence is not shown:
> 
> ![Motion](https://private-user-images.githubusercontent.com/44376749/290658758-9b365715-021c-4353-8ff2-28ed9cca4a55.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1ODkwODQsIm5iZiI6MTcwMjU4ODc4NCwicGF0aCI6Ii80NDM3Njc0OS8yOTA2NTg3NTgtOWIzNjU3MTUtMDIxYy00MzUzLThmZjItMjhlZDljY2E0YTU1LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE0VDIxMTk0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFiMDcwNzJmOTZhOTcxMzUzODg2MjhhODFmN2MxYjczNWY1MTJlZWM1NTJmMmNhM2Q2OTdiMTFlNGM1ODQyMWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.sJ47mCYwZ1zv4yQoPl7UI18HUohOH2WUUYihGaKmNcE)

Try a repair or  check the api / websocket if it works there.

hackepeterOli, 2023-12-14:
Now it works - i have to wait some minutes. Many Thanks!

hackepeterOli, 2023-12-16:
Now i can get the presence value. Is it also possible to get the occupancy value? 
That's the interesting thing about this sensor.

[image]

BabaIsYou, 2023-12-16:
Hmmm… 🤔 that’s the same ! Or is there something I didn’t understand ?

Idaho947, 2023-12-16:
The ddf works fine for me !

hackepeterOli, 2023-12-16:
Motion is triggered when the sensor detects movement - but that was not the reason why I bought this sensor. The sensor uses 5.8GHz microwave radar technology and can therefore also reliably detect people who are in a room but not moving. As far as I know, this is the occupancy value.

Idaho947, 2023-12-16:
For me it is how it work now. The sensor stay on detection even if I don t move

hackepeterOli, 2023-12-16:
Ok, maybe I need to align it even better. I'll test it again tomorrow.

BabaIsYou, 2023-12-17:
> As far as I know, this is the occupancy value.

Sorry, my answer wasn’t about semantic but based on the fact that presence item is triggered by 
            "at": "0x0000",
            "cl": "0x0406"
that is occupancy attribute you referred to …

closed 2024-01-14 by github-actions[bot]

niklas85, 2024-01-14:
Hey there! The DDF helps a lot. I got it working, but the light Sensor doesnt work for ne. Hast anybody managed to use a lux value or the dark/bright Switch from the DDF? In my case IT will stay OFF everytime...

lhering, 2024-01-14:
> Hey there! The DDF helps a lot. I got it working, but the light Sensor doesnt work for ne. Hast anybody managed to use a lux value or the dark/bright Switch from the DDF? In my case IT will stay OFF everytime...

I've added this to the general.xml:

```
    <!-- SONOFF -->
    <cluster id="0xfc11" name="Sonoff specific" mfcode="0x1286">
      <description>Cluster for light sensor</description>
      <server>
        <attribute id="0x2001" name="Light" type="u8" access="r" required="m" mfcode="0x1286">
          <description></description>
        </attribute>
      </server>
      <client>
      </client>
    </cluster>
``` 
In HA I can see the Attribute Dark with true or false but it looks like its only updating on detection.

Smanar, 2024-01-14:
>In HA I can see the Attribute Dark with true or false but it looks like its only updating on detection.

>Illumination (enum)
Only updated when occupancy is detected. Value can be found in the published state on the illumination property. To read (/get) the value publish a message to topic zigbee2mqtt/FRIENDLY_NAME/get with payload {"illumination": ""}. It's not possible to write (/set) this value. The possible values are: dim, bright

Seem the normal working mode, but you can have a value instead of just a bool value.

niklas85, 2024-01-14:
The Philips Hue motion sensors are offering a light value (lux) that will be updated independent of the presence state. So it is possible to turn the light off, if the lux value is under threshold. With the sonoffs the lights will stay on, until the person leaves... Thats not so smart...
And that is what i need/want to replace the hue motion sensors with the sonoffs. Any ideas? I am not so familiar with the DDFs and configuration.

Smanar, 2024-01-14:
>i need/want to replace the hue motion sensors with the sonoffs

If I have understand, It's not possible as Sonoff don't permit the lux access all the time

Idaho947, 2024-02-08:
Arf I didn t see this ddf was non in the stable release. Maybe someyone can merche it for the next one ?

ThiemeNL, 2024-03-13:
Hi are there any plans to finish the PR #7582?

Smanar, 2024-03-14:
I have noticied the PR owner, if after somes days there is no answer can create a new one.
You can correct it's own code but you need his validation, so if if have give up, will be useless.