dresden-elektronik/deconz-rest-plugin #7374: Problem adding new IKEA trådfri (fyrtur) after upgrading to deconz 2.42.2

phuongpham, 2023-11-23:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Just updated deconz to 2.42.2
When I try to (re)add IKEA furtyr blinds. The blinds doesn't show in Phoscon App.
When I check deconz-gui I can see the device but seems like it's in unregonized state. (please see the [screenshot)[](url)]([u
[image]

### Steps to reproduce the behavior

x

### Expected behavior

Can see the device in Phoscon App and "proper" item at deconz-gui

### Environment

- Host system: RPI4
- Running method: Rasbian with Deconz-GUI
- Firmware version: (26780700)
- deCONZ version (not Home assistant Addon version!): (2.24.2)
- Device: ConBee II 
- Do you use an USB extension cable: no
- Is there any other USB or serial devices connected to the host system? NO

phuongpham, 2023-11-25:
Just downgraded Deconz to v. 2.23.01 and it works again to add fyrtur blinds.
 
https://deconz.dresden-elektronik.de/raspbian/stable/deconz-2.23.01-qt5.deb

MSL-DA, 2023-11-26:
You can add the blinds in deCONZ (2.24.2), then they will appear in Phoscon

phuongpham, 2023-11-26:
> You can add the blinds in deCONZ (2.24.2), then they will appear in Phoscon

Thanks for your feedback.
But have you tried to this after you upgraded to 2.24.2?

MSL-DA, 2023-11-26:
Yes I have. With TREDANSEN and FYRTUR

jotzet79, 2023-11-26:
> You can add the blinds in deCONZ (2.24.2), then they will appear in Phoscon

May I ask how you achieve to add / pair new devices via deCONZ GUI? Thanks in advance, J

jotzet79, 2023-11-26:
> Just downgraded Deconz to v. 2.23.01 and it works again to add fyrtur blinds.
> 
> https://deconz.dresden-elektronik.de/raspbian/stable/deconz-2.23.01-qt5.deb

Any clue how I can downgrade to v2.23.02 on ARM64 architecture? Which arch are you on?

phuongpham, 2023-11-26:
> > Just downgraded Deconz to v. 2.23.01 and it works again to add fyrtur blinds.
> > https://deconz.dresden-elektronik.de/raspbian/stable/deconz-2.23.01-qt5.deb
> 
> Any clue how I can downgrade to v2.23.02 on ARM64 architecture? Which arch are you on?
> 
> EDIT OK, I found the repo (buster in the filename misled me -> http://deconz.dresden-elektronik.de/debian/stable/deconz_2.23.02-debian-buster-stable_arm64.deb)

$ uname -m
aarch64

$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

ChrisHae, 2023-11-27:
Don't think it is a bug in deCONZ or Phoscon App. Just added a Kadrilj and a Fyrtur with deCONZ 2.24.2. With Fyrtur I also had unknown Manufacturer und Model Identifier like in your screenshot but after I clicked on another node and then on the Fyrtur node again the correct information where displayed.

Btw. I am on windows version of deCONZ.

I could control it via deCONZ. But it took a little time until it starts to respond to Rest API commands. Now it works fine.

Sometimes it just takes a while until deCONZ has read all node descriptors and clusters. When it appears in the deCONZ GUI it can also appear in the Phoscon app a few moments later also if the "Add new blind wizzard" is already closed.

closed 2023-11-27 by Mimiix