dresden-elektronik/deconz-rest-plugin #7375: Zemismart Tuya Zigbee Real-time Smart Energy Monitor SPM01-D2TZ-ZM

tomeko12, 2023-11-23:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Zemismart SPM01-D2TZ-ZM

### Manufacturer

_TZE200_bcusnqt8

### Model identifier

TS0601

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]
[image]
[image]
[image]
[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

closed 2023-12-23 by github-actions[bot]

tomeko12, 2024-10-11:
Working DDF file:
```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_bcusnqt8",
  "modelid": "TS0601",
  "vendor": "Zemismart",
  "product": "Smart Energy Sensor SPM01-D2TZ",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/_TZE200_bcusnqt8_spm01-d2tz.json",
  "subdevices": [
    {
      "type": "ZHAConsumption",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "parse": {
            "dpid": 1,
            "eval": "Item.val = Attr.val * 10;",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/power",
          "parse": {
            "dpid": 6,
            "eval": "Item.val = ((ZclFrame.at(11) << 16) | (ZclFrame.at(12) << 8) | ZclFrame.at(13));",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "ZHAPower",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "parse": {
            "dpid": 6,
            "eval": "Item.val = ((ZclFrame.at(8) << 16) | (ZclFrame.at(9) << 8) | ZclFrame.at(10));",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "parse": {
            "dpid": 6,
            "eval": "Item.val = ((ZclFrame.at(11) << 16) | (ZclFrame.at(12) << 8) | ZclFrame.at(13));",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/voltage",
          "parse": {
            "dpid": 6,
            "eval": "Item.val = ((ZclFrame.at(6) << 8) | ZclFrame.at(7)) / 10;",
            "fn": "tuya"
          },
          "default": 0
        }
      ]
    }
  ]
}
```

pdcmoreira, 2025-03-05:
Do you get correct values? I only get power and consumption sensors and power is always 0. Consumption is increasing, so I assume it's OK.