dresden-elektronik/deconz-rest-plugin #7376: Request Namron Zigbee smart plug 16A

espenkv, 2023-11-24:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Namron zigbee smart plug 16A

### Manufacturer

Namron AS

### Model identifier

4512767

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering
[image]

### Electrical Measurement
[image]

### Any other cluster of relevance/interest

###Device Temperature Configuration
[image]

/
/

FartherOut, 2023-11-28:
White plug model identifier: 4512766
Black plug model identifier:  4512767

The basic on/off capability of the plug works (via Node-RED deCONZ integration or Home Assistant deCONZ integration) but none of the advanced features like power measurements are available.
I suspect these features are the same as in the other Namron products already present in the device list so it should be rather easy to implement.

ConBee II
Gateway version 2.24.2
Firmware 26780700

BabaIsYou, 2023-11-28:
Can you please update screenshots after doing a "read" on all of them ?

espenkv, 2023-11-28:
Is this OK?

Node info
[image]
Basic
[image]
Power Configuration
[image]
On/Off
[image]
[image]
Simple Metering
[image]
[image]
Electrical Measurement
[image]
[image]
[image]
[image]
Any other cluster of relevance/interest
###Device Temperature Configuration
[image]

BabaIsYou, 2023-11-28:
Could you try that DDF ?
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["Namron AS","Namron AS"],
  "modelid": ["4512766","4512767"],
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
        },
        {
          "name": "state/on",
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0B04"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "read": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "read": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          }
        },
        {
          "name": "state/voltage",
          "read": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          }
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0702"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "read": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 0,
            "eval": "Item.val = Attr.val * 10"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

FartherOut, 2023-11-29:
I copy/pasted your config into a json and uploaded it to my docker system but maybe I am doing something wrong, because in DDF editor I only get "Failed to open /opt/deCONZ/devices/4512766_.json" when trying to load your config.
As a test, I tried to save the "default" config for the device in the editor and it can read/load that json just fine.

BabaIsYou, 2023-11-29:
Is there a reason why you discarded all the proposed values in reporting parts of the binding portion from my suggested DDF ? Unless you’re knowing what you are doing I suggest to keep suggested min, max and change values. We would adjust them later if needed.

That beeing said … adding power item to consumption sensor is easy 

```

    {
          "name": "state/power",
          "read": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          }
        },

```

for temperature. This kind of internal device temp is not displayed on standard plug as some third party app are unable to deal with it.

But also be added easily with a new temperature sensor (binding is already part of your DDF for this cluster). Will see that later also.

Regarding current, if you’re dividing you’re loosing in precision for sure. What is the unit waited by HA ? Amp, mAmp ? What is the Amp of the device you plugged onto that smart plug ?

FartherOut, 2023-11-29:
> Is there a reason why you discarded all the proposed values in reporting parts of the binding portion from my suggested DDF ? Unless you’re knowing what you are doing I suggest to keep suggested min, max and change values. We would adjust them later if needed.

No reason at all, except I'm completely new to this and don't know what I'm doing. :)
I'm sure your code is more correct.

> Regarding current, if you’re dividing you’re loosing in precision for sure. What is the unit waited by HA ? Amp, mAmp ? What is the Amp of the device you plugged onto that smart plug ?

Indeed, I did some quick test and the plug reports mA, and in my case HA doesn't know that so it defaults to showing A instead.

BabaIsYou, 2023-11-29:
Then I suggest this DDF : 
```
{
    "schema": "devcap1.schema.json",
    "manufacturername": [
        "Namron AS",
        "Namron AS"
    ],
    "modelid": [
        "4512766",
        "4512767"
    ],
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
        {
            "type": "$TYPE_SMART_PLUG",
            "restapi": "/lights",
            "uuid": [
                "$address.ext",
                "0x01"
            ],
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "state/alert"
                },
                {
                    "name": "state/on"
                },
                {
                    "name": "state/reachable"
                }
            ]
        },
        {
            "type": "$TYPE_POWER_SENSOR",
            "restapi": "/sensors",
            "uuid": [
                "$address.ext",
                "0x01",
                "0x0b04"
            ],
            "fingerprint": {
[... 277 lines cut ...]
                    "min": 1,
                    "max": 300,
                    "change": "0x00000001"
                },
                {
                    "at": "0x0508",
                    "dt": "0x21",
                    "min": 1,
                    "max": 300,
                    "change": "0x00000064"
                },
                {
                    "at": "0x050B",
                    "dt": "0x29",
                    "min": 1,
                    "max": 300,
                    "change": "0x00000001"
                }
            ]
        }
    ]
}
```

Perhaps you'll have to delete and repair your device.

FartherOut, 2023-11-29:
Yes. your new suggestion config also shows internal temperature.
However Home Assistant shows the temp as 0.41 degrees Celsius in stead of the correct 41 degrees Celsius, so it's off by a factor of ten.
I do not know if this is a HA issue or a deCONZ issue though.
[image]

As per your suggestion, I also added a power attribute in the consumption sensor since the "unkown" was annoying me.
[image]

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": [
        "Namron AS",
        "Namron AS"
    ],
    "modelid": [
        "4512766",
        "4512767"
    ],
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
        {
            "type": "$TYPE_SMART_PLUG",
            "restapi": "/lights",
            "uuid": [
                "$address.ext",
                "0x01"
            ],
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "state/alert"
                },
                {
                    "name": "state/on"
                },
                {
                    "name": "state/reachable"
                }
            ]
        },
        {
            "type": "$TYPE_POWER_SENSOR",
[... 297 lines cut ...]
                    "at": "0x0505",
                    "dt": "0x21",
                    "min": 1,
                    "max": 300,
                    "change": "0x00000001"
                },
                {
                    "at": "0x0508",
                    "dt": "0x21",
                    "min": 1,
                    "max": 300,
                    "change": "0x00000064"
                },
                {
                    "at": "0x050B",
                    "dt": "0x29",
                    "min": 1,
                    "max": 300,
                    "change": "0x00000001"
                }
            ]
        }
    ]
}
```

BabaIsYou, 2023-11-30:
> However Home Assistant shows the temp as 0.41 degrees Celsius in stead of the correct 41 degrees Celsius, so it's off by a factor of ten.

0.41 instead of 41 is for me a factor of hundred then you can change the calculation by something like : 

`"Item.val = 10 * (Attr.val + R.item('config/offset').val)"`

FartherOut, 2023-11-30:
Nope, a spike appeared again, same as before.

BabaIsYou, 2023-12-01:
Hmm I don't have any other solution yet ;-/ Seems that it's reported by the device itself.

FartherOut, 2023-12-17:
I have been using the following configuration for some weeks now and I am 99% happy.
The power spikes are still there but they don't seem to affect the power consumption stats at all.
Perhaps the plug simply don't update the stats when it is turned "off" and thus ignores the spikes.
[image]

Although the perfectionist in me is annoyed that the power usage graph includes these spikes, it doesn't really affect anything practically.
[image]

I guess this is as good as it gets for now.

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": [
        "Namron AS",
        "Namron AS"
    ],
    "modelid": [
        "4512766",
        "4512767"
    ],
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
        {
            "type": "$TYPE_SMART_PLUG",
            "restapi": "/lights",
            "uuid": [
                "$address.ext",
                "0x01"
            ],
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "state/alert"
                },
                {
                    "name": "state/on"
                },
                {
                    "name": "state/reachable"
                }
            ]
        },
[... 251 lines cut ...]
                    "at": "0x0505",
                    "dt": "0x21",
                    "min": 1,
                    "max": 300,
                    "change": "0x00000001"
                },
                {
                    "at": "0x0508",
                    "dt": "0x21",
                    "min": 1,
                    "max": 300,
                    "change": "0x00000064"
                },
                {
                    "at": "0x050B",
                    "dt": "0x29",
                    "min": 1,
                    "max": 300,
                    "change": "0x00000001"
                }
            ]
        }
    ]
}
```

closed 2024-01-15 by github-actions[bot]