dresden-elektronik/deconz-rest-plugin #7377: Cannot join ZigBee network from deCONZ GUI

jotzet79, 2023-11-24:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

 After the last update to v2.24.2 I detected that if I start the deCONZ GUI and then leave the network, I cannot join the ZigBee network / Devices anymore. Additionally I can no longer control my Fyrtur shades from Ikea via Homebridge & homebridge-hue plugin.

### Steps to reproduce the behavior

Only a complete halt of the device, detaching the RaspBee II module, waiting 15 secs, attaching the module again, and booting up my Raspberry running on Raspberry OS 12 BookWorm.

### Expected behavior

Pressing the leave button and joining the network by pressing the button again leads to the Controller joining the network immediately.

### Environment

- Host system: Raspberry Pi 
- Running method: Raspbian OS 12 Bookworm
- Firmware version: 0x26780700
- deCONZ version (not Home assistant Addon version!): 2.24.2
- Device: RaspBee II
- Do you use an USB extension cable: not applicable
- Is there any other USB or serial devices connected to the host system? If so: External HDD

phuongpham, 2023-11-24:
Maybe similar to the problem I have?
Can you try to remove and repair the device?
I assume it could be some issue with the DDF file after the latest upgrade.
Seems like the DDF files have been updated/changed on last build
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7374

manup, 2023-11-24:
Are you sure it's not connected?
The logs show Zigbee commands being send and received:

```
12:25:24:392 APS-DATA.request id: 72, addrmode: 0x03, addr: 0x00212EFFFF06EB73, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
12:25:24:393 	asdu (length: 2): |tB��
12:25:24:435 APS-DATA.confirm id: 72, status: 0x00 SUCCESS
12:25:24:436 APS-DATA.confirm request id: 72 -> confirmed, timeout 2356494
12:25:24:481 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 232, rssi: -73
12:25:24:482 	asdu: 490002000173eb06ffff2e2100d00bc8feffe20a6824f3250101f0
```

jotzet79, 2023-11-24:
> Maybe similar to the problem I have? Can you try to remove and repair the device? I assume it could be some issue with the DDF file after the latest upgrade. Seems like the DDF files have been updated/changed on last build #7374
> 
> @ebaauw maybe you know something about this?

Yeeees, thank you! Same issues here... Oddly enough after reconnect works now, the last 3 days it didn't (what was totally frustrating)!?

So overall 2-3 problems remain -> 2 seem to be your issues too
(re)solved?
- deCONZ Internals: Cannot reconnect after leaving the network 

- Fyrtur shades no longer work via homebridge-[hue|deconz], can be controlled via deCONZ "Window Covering" Cluster properties / functions

-  Fyrtur shades can no longer be re-paired via phoscon
[image]

phuongpham, 2023-11-24:
@jotzet79 Thanks for testing this out.
So with the latest build it actually breaks re-adding fyrtur blinds.
Would be interesting if this problem also occur for other IKEA devices or any zigbee devices.

Anyway we just need to wait for some developers that can help us here :)

jotzet79, 2023-11-24:
Yeah, I agree, would be interesting if ti breaks other devices too - but then I think the number of issues would have been much higher. Anyway thanks for linking my issue to yours. I think we then could close this issue (as long as the Leave/Rejoin Network problems go away) in favour of yours.

jotzet79, 2023-11-24:
> Are you sure it's not connected? The logs show Zigbee commands being send and received:
>
> ```
> 12:25:24:392 APS-DATA.request id: 72, addrmode: 0x03, addr: 0x00212EFFFF06EB73, profile: 0x0000, cluster: [...]

Dear Manuel, yeah thanks - the initial output was copied, when I did not execute the "leave" (because I thought that this trace already gives hints if sth is wrong). However noe I have reconnect issues again:

```
23:04:07:685 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:08:264 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:08:837 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:08:898 try to reconnect to network try=6
23:04:08:929 [Master] read param with arg 0x18
23:04:08:930 [Master] read param with arg 0x13
23:04:08:930 [Master] read param with arg 0x13
23:04:08:930 [Master] read param with arg 0x13
23:04:08:962 Device protocol version: 0x010E
23:04:09:077 [Master] read param with arg 0x19
23:04:09:221 CTRL ANT_CTRL 0x03
23:04:09:237 CTRL ZDP_RESPONSE handler 0x0001
23:04:09:437 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:09:481 CTRL read param resp id: 0x13, status: 0x04
23:04:09:541 Device protocol version: 0x010E
23:04:09:964 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:10:487 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:11:015 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:11:538 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:11:898 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
23:04:12:065 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:12:588 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:12:761 [Master] read param with arg 0x18
23:04:12:762 [Master] read param with arg 0x13
23:04:12:762 [Master] read param with arg 0x13
23:04:12:762 [Master] read param with arg 0x13
23:04:12:794 Device protocol version: 0x010E
23:04:12:925 [Master] read param with arg 0x19
23:04:13:069 CTRL ANT_CTRL 0x03
23:04:13:085 CTRL ZDP_RESPONSE handler 0x0001
23:04:13:145 CTRL read param resp id: 0x13, status: 0x04
23:04:13:218 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:13:225 Device protocol version: 0x010E
23:04:13:521 [Master] read param with arg 0x18
23:04:13:522 [Master] read param with arg 0x13
23:04:13:522 [Master] read param with arg 0x13
23:04:13:522 [Master] read param with arg 0x13
23:04:13:550 Device protocol version: 0x010E
23:04:13:681 [Master] read param with arg 0x19
23:04:13:791 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:13:825 CTRL ANT_CTRL 0x03
23:04:13:841 CTRL ZDP_RESPONSE handler 0x0001
23:04:13:898 try to reconnect to network try=7
23:04:13:901 CTRL read param resp id: 0x13, status: 0x04
23:04:13:961 Device protocol version: 0x010E
23:04:14:113 [Master] read param with arg 0x18
23:04:14:114 [Master] read param with arg 0x13
23:04:14:114 [Master] read param with arg 0x13
23:04:14:115 [Master] read param with arg 0x13
23:04:14:150 Device protocol version: 0x010E
23:04:14:485 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:14:561 [Master] read param with arg 0x19
23:04:14:705 CTRL ANT_CTRL 0x03
23:04:14:721 CTRL ZDP_RESPONSE handler 0x0001
23:04:14:781 CTRL read param resp id: 0x13, status: 0x04
23:04:14:861 Device protocol version: 0x010E
23:04:15:011 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:15:054 [Master] read param with arg 0x18
23:04:15:054 [Master] read param with arg 0x13
23:04:15:054 [Master] read param with arg 0x13
23:04:15:055 [Master] read param with arg 0x13
23:04:15:082 Device protocol version: 0x010E
[... 194 lines cut ...]
23:04:38:424 [Master] read param with arg 0x13
23:04:38:501 Device protocol version: 0x010E
23:04:38:898 reconnect network failed, try later
23:04:38:899 networkState: CC_ReconnectNetwork
23:04:38:899 start reconnect to network
23:04:38:948 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:38:961 [Master] read param with arg 0x19
23:04:39:105 CTRL ANT_CTRL 0x03
23:04:39:121 CTRL ZDP_RESPONSE handler 0x0001
23:04:39:181 CTRL read param resp id: 0x13, status: 0x04
23:04:39:895 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:39:965 Device protocol version: 0x010E
23:04:40:423 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:40:949 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:41:473 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:41:897 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
23:04:41:000 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:42:524 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:43:052 poll node 58:8e:81:ff:fe:b4:63:96-01
23:04:43:898 try to reconnect to network try=1
23:04:43:963 Device protocol version: 0x010E
23:04:44:221 CTRL ANT_CTRL 0x03
23:04:44:237 CTRL ZDP_RESPONSE handler 0x0001
23:04:44:377 Device protocol version: 0x010E
```

phuongpham, 2023-11-25:
@jotzet79 
Can you verify if you get same problem after upgrading to 2.24.2?

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7380
 Please comment the results inside the issue #7380

jotzet79, 2023-11-25:
> @jotzet79 Can you verify if you get same problem after upgrading to 2.24.2?
> 
> #7380 Please comment the results inside the issue #7380

I guess this is then part of the closed source software (I guess all the GUI logic). However I am not using Desktop environment on my Raspberry.

jotzet79, 2023-12-03:
> > I guess this is then part of the closed source software (I guess all the GUI logic).
> 
> Spoiler alert: deCONZ will be open sourced fully once the cleanup is done :)
> 
> For the rejoin issue I'll do some more tests.
> 
> We have retested the Fyrtur now, initially it didn't work, the device was only visible in deCONZ and could be controlled from there via clusters. After a way too long while the configuration was done anyway and it popped up in Phoscon App and could be controlled there.
> 
> I'm not sure yet what is causing this delay.

Hi @manup, Hmm there must be some other interference now, my first guess that it has to do with the latest update was maybe wrong: also with the last firmware v2.23.2 disconnects happen, and devices become not reachable (over night). 

- Can this be connected to a weak battery (which has side effects on RTClocking, maybe)?
- Can this be connected to overlapping zigbee and wifi channels? (ZigBee Channel 15, Wifi 2,4 on Channel 8)?
- Do you have other ideas, maybe?
Thanks for pointers,
Joachim

closed 2024-01-01 by github-actions[bot]