dresden-elektronik/deconz-rest-plugin #7378: Blind with v2.24.2

MSL-DA, 2023-11-24:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Unable to add IKEA blinds

### Steps to reproduce the behavior

Add new blind

### Expected behavior

No blind found/ works with deconz:2.23.02

### Environment

Ubuntu/rpi
docker
gw-version: 2.24.2

### Additional context

works with deconz:2.23.02

Mimiix, 2023-11-24:
Closing this as it's a possible duplicate and the template was not filled properly.

closed 2023-11-24 by Mimiix

MSL-DA, 2023-11-24:
> Closing this as it's a possible duplicate and the template was not filled properly.

Why are you closing it? I don't see any error message like this and I have filled out the template as prescribed

Mimiix, 2023-11-24:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7374

MSL-DA, 2023-11-24:
> #7374

Thank you ☺️