dresden-elektronik/deconz-rest-plugin #7379: Yet another Tuya Smart Button TS004F _TZ3000_abrsvsou

x0li, 2023-11-24:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tuya Smart Scene Button

### Manufacturer

_TZ3000_abrsvsou

### Model identifier

TS004F

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

BabaIsYou, 2023-11-29:
Can you please stick to the template and give screenshots to all relevant clusters after clicking Read button ?
Thanks

x0li, 2023-11-29:
Sorry Babals, I thought I did.
Unfortunately, I'm not able to provide them anymore, as I moved all my Zigbee network to ZHA.
If it can't be fixed with the provided info, please feel free to close the issue.

Thanks for your efforts!

closed 2023-11-29 by Mimiix