dresden-elektronik/deconz-rest-plugin #7380: Deconz gui icon/shortcut missing on 2.24.2

phuongpham, 2023-11-25:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Shortcut/icon is missing from the menu on Rasbian after upgrading to deconz 2.24.2
[image]

On latest version (2.24.2) I have to start deconz-gui thru terminal. (sudo service deconz-gui start)

Works fine again if I downgrade to 2.23.01

### Steps to reproduce the behavior

1. Click on Rasbian menu
2. Go to development
3. The icon/shortcut is missing

### Expected behavior

The icon/shortcut should be visible in the menu

### Environment

Host system: RPI4
Running method: Rasbian with Deconz-GUI
Firmware version: (26780700)
deCONZ version (not Home assistant Addon version!): (2.24.2)
Device: ConBee II
Do you use an USB extension cable: no
Is there any other USB or serial devices connected to the host system? NO

SwoopX, 2023-11-25:
This has nothing to do with the REST API.

closed 2023-11-25 by SwoopX

phuongpham, 2023-11-25:
@SwoopX which repo does this belongs to?
Something changes during the 2.42.2

SwoopX, 2023-11-25:
You may raise this in the Phoscon forum.