dresden-elektronik/deconz-rest-plugin #7381: Lidl Silvercrest _TZ3000_j1v25l17 does not show energy consumption

Lucleonhart, 2023-11-26:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I purchased several _TZ3000_j1v25l17 Plugs, but they do not display the energy consumption.
I found this closed issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6995 together with the matching PR and merge https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7013

i updated all software and repaired the device but its still the same.

i am running deconz headless with homebridge. Which data is required for debugging?

### Steps to reproduce the behavior

Pair _TZ3000_j1v25l17 to deconz, check device in Eve, see no energy data.

### Expected behavior

Show energy consumption like on other _TZ3000 devices.

### Screenshots

[image]
[image]

### Environment

- Host system: (Raspberry Pi)
- Running method: (Raspbian)
- Firmware version: (26780700)
- deCONZ version (not Home assistant Addon version!): (2.24.2)
- Device: (ConBee II)
- Do you use an USB extension cable: (yes)
- Is there any other USB or serial devices connected to the host system? No

Mimiix, 2023-11-26:
Can you share a screenshot of the basic cluster In deconz ?

Lucleonhart, 2023-11-26:
How do i do that on a headless running raspberry pi?

Mimiix, 2023-11-26:
> How do i do that on a headless running raspberry pi?

You can't. Other than vnc perhaps.

Can you show a screenshot of the device in phoscon ? Sometimes it shows there too.

closed 2023-11-27 by Mimiix

unstressable, 2024-01-12:
same Problem here!:
Green ones show Energy consumption, red one not!

[image]

Mimiix, 2024-01-12:
> same Problem here!: Green ones show Energy consumption, red one not!
> 
> <img alt="image" width="629" src="https://private-user-images.githubusercontent.com/50423935/296355290-fdbdc3e6-31bc-4d4a-83d4-8bcc9e7afa8d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUwOTQ1MTUsIm5iZiI6MTcwNTA5NDIxNSwicGF0aCI6Ii81MDQyMzkzNS8yOTYzNTUyOTAtZmRiZGMzZTYtMzFiYy00ZDRhLTgzZDQtOGJjYzllN2FmYThkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEyVDIxMTY1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNmZTI3NDM3M2YxNjg2M2VkN2JmNDEwN2MwYWJkMWNjMzVjZjliYzMyMWRhZTc2N2M3MzI3MzdhYzhiZDM1N2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.3SWfD53ee7_H2IFz1AOnhg9DQzN0UPaYBW7yPbIMK4E">
> What do I have to do? ->Deconz Headless/ioBroker Phoscon App

Change the DDF to gold.

This is not the place to ask this. For help please ask on the forums.