dresden-elektronik/deconz-rest-plugin #7382: Aqara Single Switch Module T1 (With Neutral)

MSL-DA, 2023-11-26:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

It's not a bug, more of a wonder that a "switch" has been added to the device and can't be configured for anything. And the device cannot be recognized by Homebridge deCONZ. Found this answer from ebaauw https://github.com/ebaauw/homebridge-deconz/issues/137#issuecomment-1518686301

### Steps to reproduce the behavior

Nothing to reproduce

### Expected behavior

No ZHASwitch

### Screenshots

[image]

[image]

[image]

### Environment

Not relevant

SwoopX, 2023-11-26:
What else to say...

[image]

closed 2023-11-26 by SwoopX

MSL-DA, 2023-11-26:
> What else to say...
> 
> ![grafik](https://private-user-images.githubusercontent.com/4005212/285652640-24f04ffd-4181-41bf-bae1-28431d7f506f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEwMDUxMjgsIm5iZiI6MTcwMTAwNDgyOCwicGF0aCI6Ii80MDA1MjEyLzI4NTY1MjY0MC0yNGYwNGZmZC00MTgxLTQxYmYtYmFlMS0yODQzMWQ3ZjUwNmYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTEyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzExMjZUMTMyMDI4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzBhYTRkMGU0NjhkMDQxNGRkODAxMWM2YjhiM2JhMjU0MGU2Mzg1ZDRjMWZhNDI4YjY2NWNmODc0NTE4Mjk1NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.l7QQ2Uj-SNrFRycoyqnm4oVsol2N4uJ8KLaQcpMcgGA)

The right way for better support and functionality. Well done! 👍🏻

Mimiix, 2023-11-26:
> The right way for better support and functionality. Well done! 👍🏻

That's what the forum is for.