dresden-elektronik/deconz-rest-plugin #7383: Aqara Single Switch Module T1 (with neutral) SSM-U01

MSL-DA, 2023-11-26:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

There is no switch associated with this device  

```
 {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x29",
        "0x0012"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0100",
        "endpoint": "0x29",
        "in": [
          "0x0012"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/clickmode",
          "refresh.interval": 300,
          "read": {
            "at": "0x000a",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl:attr",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x000a",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'rocker') { 1 } else if (Item.val == 'momentary') { 2 } else { 'unknown' }",
            "fn": "zcl:attr",
            "mf": "0x115f"
          },
          "parse": {
            "at": "0x000a",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'rocker' } else if (Attr.val == 2) { Item.val = 'momentary' } else { Item.val = 'unknown' }",
            "fn": "zcl:attr",
            "mf": "0x115f"
          },
          "default": "rocker"
        },
        {
          "name": "config/devicemode",
          "refresh.interval": 300,
          "read": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl:attr",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'compatibility') { 1 } else if (Item.val == 'zigbee') { 2 } else { 'unknown' }",
            "fn": "zcl:attr",
            "mf": "0x115f"
          },
          "parse": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'compatibility' } else if (Attr.val == 2) { Item.val = 'zigbee' } else { Item.val = 'unknown' }",
            "fn": "zcl:attr",
            "mf": "0x115f"
          },
          "default": "compatibility"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
```

### Steps to reproduce the behavior

Nothing to reproduce

### Expected behavior

I don't expect a switch on this unit (SSM-U01)

### Environment

Not relevant

SwoopX, 2023-11-26:
What you provide as bug description is telling us the exact opposite, I'm afraid. This code is responsible for actually creating ZHASwitch resources.

Please provide some more appropriate evidence to support your report, as it it not comprehensible (yet).

MSL-DA, 2023-11-26:
You are creating a virtual switch that can't do anything. It does not exist. It is published as follows: Smart Switch WXKG11LM Rev.2 in Phoscon and Homebridge deCONZ. Search for WXKG11LM Rev.2 and Aqara Single Switch Module T1 (with neutral)

[image]

[image]

MSL-DA, 2023-11-26:
https://github.com/ebaauw/homebridge-deconz/issues/137#issue-1678507275

SwoopX, 2023-11-26:
I would strongly suggest you read all the information which is provided by our issue templates, as this falls into the categories/examples as described in the supplementary information. Please use the Phsocon forum for your case.
If you want to contribute to improve the templates, please also mention it there.

closed 2023-11-26 by SwoopX