dresden-elektronik/deconz-rest-plugin #7386: Frient Keypad : adding more codes than just master code to alarm system in order to have RFID tags working

Gw3n4, 2023-11-26:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Frient keypad KEPZB-110

### Manufacturer

Frient

### Model identifier

KEPZB-110

### Device type to add

Other

### Node info

[image]
[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Gw3n4, 2023-11-26:
OK, thank you. I didn't know i could comment on the previous issue as it was closed.

Mimiix, 2023-11-26:
You can unless it's locked :)!

closed 2023-11-26 by Mimiix