dresden-elektronik/deconz-rest-plugin #7387: Clickmode Aqara Wireless Remote Switch (WRS-R02)

Harry423, 2023-11-26:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

The attribute "clickmode" can not be changed. The should set to "multiclick". I've change the value (via deCONZ adapter in ioBroker) - but the events don't come right.
Normaly you can detect single, double and tripple click and long click for each button. For this you get different value at the "buttonpressed" attribute. 
After changing the value "clickmode" always only 1002 or 2002 would be detected. Like before the changing den "clickmode" attribute.

[image]

The attributes "double_press" and "tripple_press" are not created.
I have an another switch (the same model) that I've added one version (2.24.0) earlier, this switch work well in "mulitclick" mode. 

[image]

How can I change this mode correctly?

### Steps to reproduce the behavior

I've deleted the switch at the phoscon app and at ioBroker too. After this I've added the switch again. Twice I've tried to reprogram the switch with the original Aqara gateway M1S first to multiclick mode and add the switch after that procedure (and deleting from the Aquaria app) to Phoscon again.

### Expected behavior

If I change the "clickmode" should be changed like at the original gateway.

### Environment

- Host system: Raspberry Pi 3B
- Running method: Raspbian
- Firmware version: 26780700
- deCONZ version (not Home assistant Addon version!): 2.24.2
- Device: RaspBee II)
- Is there any other USB or serial devices connected to the host system? If so: No.

SwoopX, 2023-11-26:
Issues with iobroker should be reported in the respecive repository/community.
Setting any configuration with the original gateway will only work as long as the device is connected to it. By switching the gateway, devices are always reset. Setting the mode via deconz works.

closed 2023-11-26 by SwoopX

Harry423, 2023-12-09:
Hello SwoopX, thank you for your quick response. But I have to apologize, I'm probably being a bit too stupid.
How can I set this via deconz? I tried to do it with the graphical user interface and failed miserably. I have this running on a Raps 3+ and have logged on to the graphical interface as pi. I get deconz to start, but I can't connect to the Conbee stick. But I can see the node list. If I log in as root I can switch on the stick, but I can no longer see the nodes from the database. I have of course added pi to the "dailout" group, as stated in the documentation.  
Can I perhaps also set this with the shell? Unfortunately, I haven't found anything in the graphical user interface.
Thanks for your help.