dresden-elektronik/deconz-rest-plugin #7390: Correct support for Bticino K4027C rolling shutter switch (% not working outside Deconz Gui)

mariomaz87, 2023-11-27:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Bticino Living Now Rolling Shutter Switch

### Manufacturer

Bticino

### Model identifier

K4027C

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

Mimiix, 2023-11-27:
Please add all screenshots.

mariomaz87, 2023-11-27:
> Please add all screenshots.

Hi, I added the Window Covering, Legrand Specific and Binary Input. Let me know if you need anything else!
As commented previously, with the current implementation I can control everything using Deconz Gui, while in Phoscon and Home Assistant I can open, close, stop and it will report the correct %, but I can't set it to a specific lift % as I can in Deconz Gui. Thanks!

Mimiix, 2023-11-27:
Looks good :)

@smanar can you help out ?

Smanar, 2023-11-27:
I m searching the problem on code, but I don't see it.
And we can't use DDF for covering, no support for state/open and state/lift for action.

You haven't an error message as return ? You can to use command line to have it.

`curl -H 'Content-Type: application/json' -X PUT -d '{"lift": 50}' http://IP:PORT/api/KEY/light/ID/state`

IP and PORT are the same used in phoscon.
KEY is an api key, can use the same than in HA
ID is the device ID.

Smanar, 2023-11-27:
BTW, when you use the GUI to move it, you are using the "Go to Lift Percentage" command ?

mariomaz87, 2023-11-27:
> BTW, when you use the GUI to move it, you are using the "Go to Lift Percentage" command ?
> 
> Because the code can reverse it but not cancel it ....
> 
> Edit: Ha yes I remember now, I think I have found, but it's not a good news

Yes, I use that command. No good news?

Smanar, 2023-11-27:
I was searching the original issue.
I just find it https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3532#issuecomment-1251038883

It's because first covering version was not supporting lift, so there is a hack in the code.

I have fast checked the code and for me it still not possible using DDF, I m asking to other devs to be sure.
Are you able to compile a new c++ code if I give you the request to use ? (you need a linux machne with full OS)

mariomaz87, 2023-11-27:
> I was searching the original issue. I just find it [#3532 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3532#issuecomment-1251038883)
> 
> It's because first covering version was not supporting lift, so there is a hack in the code.
> 
> I have fast checked the code and for me it still not possible using DDF, I m asking to other devs to be sure. Are you able to compile a new c++ code if I give you the request to use ? (you need a linux machne with full OS)

Deconz is running on Raspbian on a Pi4, so I think I can compile!

mariomaz87, 2023-11-27:
> I m searching the problem on code, but I don't see something special (but there is so much hack for this device) And we can't use DDF for covering, no support for state/open and state/lift for action.
> 
> You haven't an error message as return ? You can to use command line to have it.
> 
> `curl -H 'Content-Type: application/json' -X PUT -d '{"lift":50}' http://IP:PORT/api/KEY/light/ID/state`
> 
> IP and PORT are the same used in phoscon. KEY is an api key, can use the same than in HA ID is the device ID.
> 
> ```
> "4":{
>    "config":{
>       "groups":[
>          "0"
>       ]
>    },
>    "etag":"58fa2315a5226f4996e1a75bf8d7659c",
>    "hascolor":false,
>    "lastannounced":null,
>    "lastseen":"2023-11-27T16:29Z",
>    "manufacturername":"Legrand",
>    "modelid":"Shutter SW with level control",
>    "name":"Finestra Bagno Vasca",
>    "state":{
>       "bri":0,
>       "lift":0,
>       "on":false,
>       "open":true,
>       "reachable":true
>    },
>    "swversion":"0x003643FF",
>    "type":"Window covering device",
>    "uniqueid":"00:04:74:00:00:9c:4c:ed-01"
> }
> ```

Will it still help to see the error message? I don't know how to find the API key in HA, but I can look it into if you need to troubleshoot the issue!

Smanar, 2023-11-28:
>Deconz is running on Raspbian on a Pi4, so I think I can compile!

Nice, so we can make a try.
I m using this code

```
        else if (taskRef.lightNode->modelId() == QLatin1String("Shutter switch with neutral") ||
                 taskRef.lightNode->modelId() == QLatin1String("Shutter SW with level control"))
        {
            // Legrand invert bri and don't support other value than 0
            bool bStatus = false;
            uint nHex = taskRef.lightNode->swBuildId().toUInt(&bStatus, 16);
            if (bStatus && nHex < 28)
            {
                targetLiftZigBee = targetLift == 0 ? 100 : 0;
            }
            else if (bStatus && nHex < 44)
            {
                targetLiftZigBee = targetLift == 100 ? 100 : 0;
            }
            else
            {
                // New devices no need thoses previous hack
                targetLiftZigBee = targetLift;
            }
        }
```

As you have `"swversion":"0x003643FF",` I think the hack will be skipped, probably because the value is too big to be converted to a Int.

So to test this code

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz (The code is based on realy last one, so need a recent version):

```
sudo apt install deconz-dev
git clone --branch legrand_10 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

It don't impact your zigbee database, but can use a lot the CPU (at least the first time, compilation will be faster after).
To rollback fastly you can make a backup of the file /usr/share/deCONZ/plugins/libde_rest_plugin.so in another folder, but thoses command will change only this file.

mariomaz87, 2023-11-29:
#2811 this is the issue I'm having but I couldn't find a solution, thanks!

Smanar, 2023-11-29:
You have done the step 2 and 4 ? from here https://phoscon.de/en/conbee2/install#raspbian
You have updated your OS recently ?

```
sudo apt update
sudo apt upgrade
```

mariomaz87, 2023-11-29:
> You have done the step 2 and 4 ? from here https://phoscon.de/en/conbee2/install#raspbian You have updated your OS recently ?
> 
> ```
> sudo apt update
> sudo apt upgrade
> ```

Yes, and I can install the deconz package alone. It’s the deconz-dev the problem

Smanar, 2023-11-29:
What is your Raspian version ?
There is perhaps a package version issue.

```
pi@raspberrypi:~ $ grep PRETTY_NAME /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
pi@raspberrypi:~ $ cat /etc/apt/sources.list.d/deconz.list
deb http://phoscon.de/apt/deconz buster-beta main
pi@raspberrypi:~ $ sudo apt install deconz-dev
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances
Lecture des informations d'état... Fait
deconz-dev est déjà la version la plus récente (2.22.00-raspbian-buster-beta).
```

mariomaz87, 2023-11-29:
> What is your Raspian version ? There is perhaps a package version issue.
>
> ```
> pi@raspberrypi:~ $ grep PRETTY_NAME /etc/os-release
> PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
> pi@raspberrypi:~ $ cat [...]

I’m on Raspian 11, bullseye

mariomaz87, 2023-11-30:
Thanks, in the mean time I have installed on my Rasp all the packages required for CMake

Smanar, 2023-12-01:
I m on it ATM, still not finished, my pi3 is not able to compile it for the moment, it freeze, I need to stop it with (ctrl+C) and restart (up the next freeze). But for the moment the request are

```
sudo apt-get update && sudo apt-get install --no-install-recommends -y lsb-release ca-certificates build-essential pkg-config git libqt5serialport5-dev libqt5websockets5-dev qtdeclarative5-dev sqlite3 libsqlite3-dev libgpiod-dev libssl-dev curl cmake ninja-build
git clone --branch legrand_10 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
cmake -DCMAKE_INSTALL_PREFIX=/usr -G Ninja -B build
cmake --build build
cmake --install build --prefix tmp
sudo cp tmp/share/deCONZ/plugins/libde_rest_plugin.so /usr/share/deCONZ/plugins
```

mariomaz87, 2023-12-01:
> I m on it ATM, still not finished, my pi3 is not able to compile it for the moment, it freeze, I need to stop it with (ctrl+C) and restart (up the next freeze). But for the moment the requests are
> 
> ```
> sudo apt-get update && sudo apt-get install --no-install-recommends -y lsb-release ca-certificates build-essential pkg-config git libqt5serialport5-dev libqt5websockets5-dev qtdeclarative5-dev sqlite3 libsqlite3-dev libgpiod-dev libssl-dev curl cmake ninja-build
> git clone --branch legrand_10 https://github.com/Smanar/deconz-rest-plugin.git
> cd deconz-rest-plugin
> cmake -DCMAKE_INSTALL_PREFIX=/usr -G Ninja -B build
> cmake --build build
> cmake --install build --prefix tmp
> sudo cp tmp/share/deCONZ/plugins/libde_rest_plugin.so /usr/share/deCONZ/plugins
> ```

It worked, thanks! I compiled it in under 5 minutes with my Pi4 and after restarting deconz I can now set a specific % from HomeKit. 
Should I reapply this plugin every time deconz is updated? Thank you so much!

Smanar, 2023-12-01:
>It worked, thanks! I compiled it in under 5 minutes with my Pi4 and after restarting deconz I can now set a specific % from HomeKit.

Seriously ? so fastly ? On first try ?
I m trying too on windows now, and I give up too, too much missing request.

>Should I reapply this plugin every time deconz is updated? Thank you so much!

If it work for you I can submit a PR to add this change officialy, hoping for it will be merged.

mariomaz87, 2023-12-01:
> > It worked, thanks! I compiled it in under 5 minutes with my Pi4 and after restarting deconz I can now set a specific % from HomeKit.
> 
> Seriously ? so fastly ? On first try ? I m trying too on windows now, and I give up too, too much missing request.
> 
> > Should I reapply this plugin every time deconz is updated? Thank you so much!
> 
> If it work for you I can submit a PR to add this change officialy, hoping for it will be merged.
> 
> So all is working for you ?

So far everything is working as expected. I can test it for a few days and get back to you if everything is ok?

Smanar, 2023-12-01:
>So far everything is working as expected. I can test it for a few days and get back to you if everything is ok?

Sure.
Have made a PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7410 will see Dev reaction (hard to purpose c++ code modification ATM)

Smanar, 2023-12-01:
@mariomaz87 and BTW thx for your help ^^, On my side I m still trying to compile my own code.

Smanar, 2023-12-02:
Yes, because we probably need to update the "How to" so your tests are realy usefull.
On my side I had other problems not specially related to it ^^.

mariomaz87, 2023-12-07:
Hi, after extensive testing I noticed two things:
- the % is inverted only in Phoscon app. In HomeKit and Home Assistant is 20% while in Phoscon is 80%. The correct one is 20%. 
- Alexa discovers the device but I can’t control it. It’s not discovered as blind but as “other” and there are no commands for it. 

Thanks!

ebaauw, 2023-12-07:
> the % is inverted only in Phoscon app. In HomeKit and Home Assistant is 20% while in Phoscon is 80%. The correct one is 20%.

That is conform spec: Zigbee uses % closed where HomeKit uses % open.  deCONZ and Phoscon follow Zigbee.

mariomaz87, 2023-12-08:
> Case closed in that case :)

I think we are waiting for PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7410 to fix the support for the device, thanks!

Mimiix, 2023-12-08:
Ah wait your right!

My apologies.

Smanar, 2023-12-08:
And for alexa, idk, the json seem fine for me

`"type":"Window covering device",`

You have a working one to compare ?

mariomaz87, 2023-12-08:
> And for alexa, idk, the json seem fine for me
> 
> `"type":"Window covering device",`
> 
> You have a working one to compare ? But If I m right Alexa don't use open and lift ? I have realy no idea how it work for covering.

I have other blinds in Alexa and they are showing as “window blinds”. 
I’m having issues with Deconz native integration with Alexa, my devices are not responding and if I remove them I can’t add them again because Alexa can’t discover them. I manage to integrate them back in Alexa using the homebridge-Alexa plugin and also the blind object of this thread is working as expected.

Smanar, 2023-12-09:
>I have other blinds in Alexa and they are showing as “window blinds”.

And they are on deconz too ? Can you share the device json to compare ?

>Maybe it’s because I changed the port of Deconz, I have to try to fix this issue

This ins invisible for Alexa, it can only see data returned by the API.

mariomaz87, 2023-12-09:
> > I have other blinds in Alexa and they are showing as “window blinds”.
> 
> And they are on deconz too ? Can you share the device json to compare ?
> 
> > Maybe it’s because I changed the port of Deconz, I have to try to fix this issue
> 
> This is invisible for Alexa, it can only see data returned by the API.

The working blinds are not using Deconz so I don’t have a json to compare. 
But right now my whole Alexa\Deconz is broken. Alexa can’t discover anything.

Smanar, 2023-12-09:
>The working blinds are not using Deconz

And by curisosity what is the application they are using ?

mariomaz87, 2023-12-09:
Control by Bticino\Legrand using Bticino Gateway

mariomaz87, 2023-12-11:
The problem with Alexa not discovering the devices is related to Deconz port. If I switch it back to 80, it discovers the devices. The lights work and the blind is unresponsive.

Smanar, 2023-12-11:
Haaa, yes I remember now, long time I have give up for Alexa stuff, and I don't found the "tutorial for Alexa" on the wiki (only this link from 2018 https://github.com/dresden-elektronik/deconz-rest-plugin/issues/759#issuecomment-417944535)
If I m remember there is too skill designed to work with some third app, and thoses one can support deconz like that https://www.npmjs.com/package/homebridge-alexa

Deconz mimic the HUE Gateway (at the start, the more time passes, the further away it gets), and I m not sure this one support covering

mariomaz87, 2023-12-11:
> Haaa, yes I remember now, long time I have give up for Alexa stuff, and I don't found the "tutorial for Alexa" on the wiki (only this link from 2018 [#759 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/759#issuecomment-417944535)) If I m remember there is too skill designed to work with some third app, and thoses one can support deconz like that https://www.npmjs.com/package/homebridge-alexa
> 
> Deconz mimic the HUE Gateway (at the start, the more time passes, the further away it gets), and I m not sure this one support covering

Yes, I'm asking for support in the Homebridge Alexa repository because I have the same exact problem also using that plugin (the shutter switch is discovered but it's not reachable). Thanks anyway!

closed 2023-12-14 by manup