dresden-elektronik/deconz-rest-plugin #7394: Gold Status for DDF of #6807

GabeHH78, 2023-11-27:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Hi,

could you pls put the DDF of #6807 to Gold Status, so that it works with Phoscon App.

Thanks

### Steps to reproduce the behavior

Gold Status for DDF #6807

### Expected behavior

Gold Status for DDF #6807

### Environment

- Host system: (Raspberry Pi / PC / NAS / MAC)
- Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
- Firmware version: (26xxyy00)
- deCONZ version (not Home assistant Addon version!): (2.xx.yy)
- Device: (ConBee I / ConBee II / ConBee III / RaspBee I / RaspBee II)
- Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?

GabeHH78, 2023-11-28:
@SwoopX could you pls inform why it is closed? Unfortunately the DDF not function with Phoscon App in silver state and a pull request not function. So how I should handle it???

SwoopX, 2023-11-28:
Ah, sure. You have acknowledged you want to report a real bug, which this is clearly not and therefore, the issue was closed. We assume the supplementary information provided while creating an issue here gives very clear guidance on what to do in which case? If not, please let us know how to make things better 👍