dresden-elektronik/deconz-rest-plugin #7398: Innr RB 286 C turns on randomly

MaStahl84, 2023-11-28:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I have two new Innr RB 286 C in my network. The lamps seems to work fine at the first moment, but have a strange behavior.
Sometimes these two Lamps turn both on, always at the same time, when an other group is switched on. It doesn't matter which group. Even when the lamps are not in a group, both turn on.

I tried to reset the Lamps, made an OTA-Firmware update, but the problem still exists.

The lamps are named "Terasse Wandlampe 1" and "Terasse Wandlampe 2"

### Steps to reproduce the behavior

1. add an innr RB 286 C to the Network, put them in any group. 
2. turn any group in your network on and off, repeat this 5,6 or 7 times
3. the innr lamps go on.

### Expected behavior

Ligth stays off

### Environment

- Host system: Synology NAS
- Running method: Ubuntu Virtual Machine
- Firmware version: 264D0900
- deCONZ version : (2.24.2)
- Device: ConBee III
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? If so: no

### deCONZ Logs

[Deconz Log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13493297/Deconz.Log.txt)

BabaIsYou, 2023-11-29:
Did you have a look at this ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/What-you-can-do-when-a-switch-turns-(almost)-all-devices-on-off

closed 2023-12-29 by github-actions[bot]

MaStahl84, 2024-01-12:
Hello @BabaIsYou ,
thank you for your answer. Unfortunatly i coud not write earlier.

The Lights are not switched on with a switch. I directly call a service in Homeassistant from nodered to turn my light groups on or off.
In my setup there are over 90 lights an 40 groups. An all other lights work as expected.

The ight (innr RBC286 C) with the problem ist in a group with other lights. Now when i turn on any other group in NoderRed with a service call in HomeAssistant, only this light turns sometimes on.. It is equal in witch group this light is assigned to, also other lights from the same type are switching on at the same moment, even if they are in different groups.
I tried to reset the bulbs an reconnected them to the network, but the behavior is the same. I also updated the bulbs with the newest firmware.
Bulbs from other manufactures (Paulmann), witch i connected after the innr bulbs, don't have this problem.

This behavior can easly be reproduced:
I connect ab new innr RB 286 C to the zigbee network, then i call a service api with the parameter configure for groups to "on", 5 or 6 times -> light goes on.

When logs are needed, please let me know witch and i will attach them here.

Thank you

MaStahl84, 2024-02-12:
I could solve the Problem with an ota firmware update of the bulbs.