dresden-elektronik/deconz-rest-plugin #7411: Bosch Water Alarm / Sensor BSEN-W

Schaussi, 2023-12-01:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Smart Home Wassermelder / Water alarm

### Manufacturer

Bosch

### Model identifier

unknown

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

Unfortunately, no more information visible yet.

### Basic

Unfortunately, no more information visible yet.

### Further relevant clusters

Unfortunately, no more information visible yet.

The device seems to be similar to the BWA-1 model, but has "BSEN-W" written on the bottom side.
So I am not sure if it's a duplicate of e.g. #5190 - if so, please delete my request.

I tried pairing via installcode as described [here](https://forum.phoscon.de/t/bosch-water-sensor-bwa-1-how-to-set-up-using-install-code/1698/10), which did not work (device not found in Phoscon web UI, is at least visible on the deCONZ map, but with only few informations.
Please give me some advice if there is a way to make the BSEN-W usable with deCONZ.

SwoopX, 2023-12-01:
Hi there,

please pair the device while using the install code printed on the device (https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/devices/#pair-with-install-code). This enables you to share the missing information

Schaussi, 2023-12-01:
Great, it finally worked.

Some notes maybe useful for others:
- If you scan the QR code, the "installcode" are the _last_ 36 characters of the scanned string
- I used HA with deconz.configure to send the installcode, until I found out this was not working, "no master gateway available" in logs. Now I used Chrome/RestMan, aquired an API key and could post the installcode successfully.

The sensor was then found via usual web UI sensor scan.
It has one entity (battery) yet, I will send you the missing deCONZ screenshots tomorrow, which are now visible.
Thank you so far.

SwoopX, 2023-12-02:
No further to do, it's indeed a duplicate of #5190

closed 2023-12-02 by SwoopX