dresden-elektronik/deconz-rest-plugin #7413: Raspberry Pi 5 + Raspbee = no devices found

00lex, 2023-12-04:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Hi! i was running deconz on my rpi4 with raspbee2 previously without issues. 
Now i swapped the sd card and the module to the rpi5. 
- in phoscon firmware is not connected. 
- with "GCFFlasher_internal -l" I get "no devices found"

is deconz not ready for rpi5 or do I have a GPIO issue

### Steps to reproduce the behavior

help

### Expected behavior

help

### Environment

- Host system: Raspberry Pi 5
- Running method: Raspbian 12 64
- Firmware version: latest 
- deCONZ version: latest stable + latest beta
- Device: raspbee2 
- Do you use an USB extension cable: no because raspbee2 + sd-card
- Is there any other USB or serial devices connected to the host system? no

Mimiix, 2023-12-05:
Great you figured it out :)

Thanks for sharing!

I'll close when Dresden updates the install guide.

Mimiix, 2023-12-05:
To make a note: RPI5 is not listed as supported models on Phoscon docs as of now:

https://phoscon.de/en/raspbee2/install#raspbian

ebaauw, 2023-12-05:
You probably want to use a GPIO riser.  I've been using these for the RaspBee II on my Raspberry Pi 4B with heatsink case.  Make sure to get one with medium to long pins; the RaspBee doesn't make good contact with the shorter variants.

The Pi 5 has a separate header for the serial port and no longer exposes it on GPIO 14 and 15, where the RaspBee expects it.  
The [documentation](https://datasheets.raspberrypi.com/rp1/rp1-peripherals.pdf) of the RP1 south bridge chip states that it supports up to 5 UARTs, that can be configured using alternate GPIO functions.  I think adding the following line to `/boot/config.txt` should expose UART0 on GPIOs 14 and 15:
```
gpio=14,15=a4
```

See section 3.1 of the documentation and https://www.raspberrypi.com/documentation/computers/config_txt.html#gpio-control

00lex, 2023-12-05:
> Hi we received our first RPi5 today to make some tests and update the docs. The (optional?) heat sink is a bit problematic as it sits right where RaspBee is located.

Maybe but the rpi5 runs in my environmental 2 degree Celsius cooler. I'll leave the heat sink in its box for now

00lex, 2023-12-05:
just to finish this. sry off-topic. 

- GPIO riser OR a lot male/female jumper wire. that's by the way good for distance when using USB3 on rpi4

- and here it comes! in a short 15 minute test with USB3, there were no problems with the zigbee network. tested with rsync from SSD and full speed to my laptop. with the RPI4 without GPIO extensions, this was the death blow for zigbee. 
if this turns out to be true, it would be the best innovation after PCI-E and there would only be one enemy left. 2.4ghz wifi ^^

00lex, 2023-12-12:
> You probably want to use a GPIO riser. I've been using these for the RaspBee II on my Raspberry Pi 4B with heatsink case. Make sure to get one with medium to long pins; the RaspBee doesn't make good contact with the shorter variants.

or something like that :):
[image]

the reliability with usb3 without distance to the zigbee module is much much better. with the rpi4 I was able to disconnect the zigbee network with a lot of load on usb3. with the rpi5 without extension there are only rare dropouts. however. if you don't want to have stress, you continue to use extensions

00lex, 2023-12-12:
> You probably want to use a GPIO riser. I've been using these for the RaspBee II on my Raspberry Pi 4B with heatsink case. Make sure to get one with medium to long pins; the RaspBee doesn't make good contact with the shorter variants.

or something like that :):
[image]

the reliability with usb3 without distance to the zigbee module is much much better. with the rpi4 I was able to disconnect the zigbee network with a lot of load on usb3. with the rpi5 without extension there are only rare dropouts. however. if you don't want to have stress, you continue to use extensions 

btt:
this wasnt a bug. the solution is: 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7413#issuecomment-1840734778

case closed

closed 2023-12-12 by 00lex

ebaauw, 2023-12-12:
Still cannot get my Pi 5 to recignise my old RaspBee (original model).  @00lex what is your entire `/boot/config.txt`?  And I assume you did the no serial console, but enable serial port from `raspi-config`?

I now have:
```
dtparam=uart0_console
dtparam=uart0=on
```
I edited the first line as per above suggestion, and I think `raspi-config` added the second line.  The GPIOs are configured for the serial port:
```
$ pinctrl get 14-17
14: a4    pn | hi // PIN8/GPIO14 = TXD0
15: a4    pu | hi // PIN10/GPIO15 = RXD0
16: no    pd | -- // PIN36/GPIO16 = none
17: no    pd | -- // PIN11/GPIO17 = none
```
But the RaspBee isn't recognised by GCFFlasher_internal, nor by deCONZ.

00lex, 2023-12-12:
# deconz
enable_uart=1
dtparam=uart0_console

ebaauw, 2023-12-14:
OK, Apparently I managed to brick one my old RaspBee devices.  The other one is working with these settings, though.

nsk, 2023-12-26:
Hi I tested your solution, no change for me I'm going to wire my rasbee2 in serial usb to see if I can flash a firmware directly to see if that solves anything, but no documentation on how to do it, will fumble around: pin 1 pin 6 and pin 8 for communication with the raspbee and Coolterm see if I can communicate with the raspbee ....
[image]

krebsvo, 2024-01-27:
Where can I find the /boot directory (the file config.txt)?
I installed the last relay from the Git Hub page on the RI 5.