dresden-elektronik/deconz-rest-plugin #7415: zbeacon DS01 Open/Close Sensor

traschke, 2023-12-05:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

zbeacon

### Manufacturer

zbeacon

### Model identifier

DS01

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

traschke, 2023-12-05:
I have found the following forum post which states sensor could be a Sonoff/eWeLink clone/brand: https://community.smartthings.com/t/edge-driver-mc-zigbee-drivers-for-motion-open-close-moisture-smoke-co-sensors-and-others-devices/231689/4058
The model identifier DS01 is also used by the Sonoff SNZB-04, maybe it's the same sensor in another housing.

The product page I ordered this sensor from however states it uses some kind of Tuya chip: https://a.aliexpress.com/_EG3u95B

Other than that I could not find much about this sensor.

Raupinger, 2023-12-07:
Hi there. not using dconz myself, but this was the only thing that came up when is searched for "zbeacon DS01" so here what i know: took mine apart and assuming [these](https://community.hubitat.com/t/sonoff-snzb-04-zigbee-door-window-contact-sensor-internal/47475) pictures of the sonoff sensor are accurate and current, mine looks very different. The main chip is a TLSR8258 F512ET32 In case anyone interested comes across this, here's a photo of the pcb as well as some of my notes on where the chips pins are connected:

[image]

[image]

closed 2023-12-14 by manup