dresden-elektronik/deconz-rest-plugin #7417: Ikea Parasoll

johannescramer, 2023-12-06:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Parasoll

### Manufacturer

IKEA of Sweden

### Model identifier

PARASOLL Door/Window Sensor

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### IAS Zone
Zone Status is 0x0000 when closed and 0x0001 when open. 
[image]

ebaauw, 2023-12-06:
Could you please read the attribute and re-post screenshots with the values.  Make sure to wake the device (by moving the magnet) while reading.

Not yet available here in NL, unfortunately.  They also announced a new motion sensor and a leak sensor, see https://www.ikea.com/global/en/newsroom/innovation/parasoll-badring-vallhorn-smart-sensors-231128/.

Looks like a regular _IAS Zone_ contact sensor on endpoint 0x02, which can control lights directly from endpoint 0x01.  Doesn't look like there's anything to expose from endpoint 0x01, other than the battery.  If I'm being pure, this device would be exposed as two `sensors` resources:
- A ZHAOpenClose with `uniqueid` ending in `-02-500`.  Interesting to see if the device has tampered detection, and if it does periodic and restore reports.  Otherwise, we might need an additional binding and attribute report configuration to work around that.  We would need to check whether it honours group bindings from the _On/Off_ cluster, and, if so, expose `config/group`.
- A ZHABattery with `uniqueid` ending in `-01-0001`.  Alternatively, the battery could be exposed as `config/battery` on the ZHAOpenClose resource, as is done on most other `sensors` resources (although they typically only correspond to a single endpoint).  Also need to check whether it reports battery in 0.5% (as per the standard, and with most recent IKEA firmwares) or in full % (as older IKEA firmwares).

johannescramer, 2023-12-07:
Oh, sorry. I'm new to this. I thought it had read the attributes, but apparently not all of them. 

### Basic cluster attributes
[image]

### Power Configuration
[image]

### Poll control
[image]

### Diagnostics 
[image]

### Ikea Specific
[image]

### IAS Zone Basic
[image]

### IAS Zone
[image]

I hope this is all you need. If you got any other stuff I should try, let me know. I am willing to try stuff, but have limited knowledge about deconz and zigbee in general.

closed 2023-12-14 by manup

hyper2910, 2023-12-31:
no status change via mqtt 

STATE is evrytime INITIALIZED

as Reading also only the following coming
<html>
<body>
<div class="col_header pinHeader detail_Readings" data-name="detail_Readings" style="font-family: Roboto; background: rgb(34, 34, 34); fill: rgb(34, 34, 34); height: 18px; padding: 5px 8px; color: white; border-radius: 0px; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Readings<div class="pin pinIn" style="font-family: Roboto; float: right; width: 1em; height: 1em; margin-left: 0.5em; background-image: url(&quot;data:image/svg+xml;utf8,<svg viewBox=\&quot;0 0 1792 1792\&quot; xmlns=\&quot;http://www.w3.org/2000/svg\&quot;><path fill=\&quot;rgb(128, 166, 255)\&quot; d=\&quot;M896 1088q66 0 128-15v655q0 26-19 45t-45 19h-128q-26 0-45-19t-19-45v-655q62 15 128 15zm0-1088q212 0 362 150t150 362-150 362-362 150-362-150-150-362 150-362 362-150zm0 224q14 0 23-9t9-23-9-23-23-9q-146 0-249 103t-103 249q0 14 9 23t23 9 23-9 9-23q0-119 84.5-203.5t203.5-84.5z\&quot;/></svg>&quot;); cursor: pointer; display: block;"></div></div>
IODev | deCONZ | 2023-12-31 12:46:27
-- | -- | --
attrTemplateVersion | Xiaomi_Aqara_MCCGQ11LM_Window_Door_Sensor_20211015 | 2023-12-31 13:06:25
battery | 94 | 2023-12-31 13:03:26
batteryPercent | 94 | 2023-12-31 13:03:26
reachable | 1 | 2023-12-31 13:03:26

</body>
</html>