dresden-elektronik/deconz-rest-plugin #7418: _TZ3000_mkhkxx1p / TS0001

aramcap, 2023-12-07:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Good Wife Zigbee

### Manufacturer

_TZ3000_mkhkxx1p

### Model identifier

TS0001

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering
[image]

### Electrical Measurement
[image]

### Any other cluster of relevance/interest
This appliance appears in Phoscon app, and it can be used (only on/off switch) from Home Assistant. The problem is with electrical measurement, that not appears in Phoscon app, neither Home Assistant.

Also, I'm not considered this like a light, else a plug or similar, because it's supports up to 16A and also supports energy metering. It's perfect for energy measurement of domestic appliances, and fits perfect to electrical junction boxes.

Here is the link of the product:
https://www.amazon.es/dp/B0BZ4FHV28

Thanks in advance.

aramcap, 2023-12-29:
UP

Smanar, 2023-12-30:
Hello, can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_mkhkxx1p",
  "modelid": "TS0001",
  "product": "1CH Zigbee Power Metering Switch",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300
        },
        {
          "name": "state/voltage",
          "refresh.interval": 300
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 60,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 60,
          "change": "0x00000010"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 60,
          "change": "0x00000050"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 60,
          "change": "0x00000005"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 1800
        }
      ]
    }
  ]
}
```

Values will need to be adjusted, just give the value you have and the one you need (for convertion *10 or /10 for exemple)

aramcap, 2024-01-22:
Hi. First, thanks for your response. In Phoscon app, I can see now consumption, power and voltage. Instead, in Home Assistant, only appears consumption and power, no trace of voltage. What is the problem? Can you help me, please? Thanks!

Moreover, it is the first time that I've dedicated time to read, write and understand a DDF. ¿Bindings are a requirement?

Also, I've tried to solve a DDF problem with another device (xiaomi_zncz04lm_smart_plug). The current temperature appears in 0º. I've declared a ZHATemperature object, but... some fails.

Regards!

Smanar, 2024-01-22:
>no trace of voltage

The field is not present at all ? Can be a "synchronisiation" issue, you can check direclty in the API if the field is present, using Phoscon/help/API Information/Sensors . If no the problem is from deconz, else it can happen if the field appear in the API after the sensor was detected in HA.
You don't need values convertion ?

For the plug, you are sure having the temperature and working ? (visible on the deconz GUI), because I m seing some DDF for ZNCZ04LM but without temperature.
It's possible this device use the Xiaomi cluster insteod of the temperature cluster, so in this situation you will have no and no report, all native, but I can found information on google when searching "xiaomi_zncz04lm_smart_plug"

closed 2024-02-21 by github-actions[bot]

aramcap, 2024-08-22:
Hi. Sorry for my late response.

The device _TZ3000_mkhkxx1p are run perfect in Phoscon GW (I've checked through the API and the WebUI), but in Home Assistant still doesn't work. I've done some research and I've created this issue in HA: https://github.com/home-assistant/core/issues/124427

Regarding the another device, xiaomi_zncz04lm_smart_plug, I've found the problem with the temperature sensor: the value needs to be multiplied by 100.

I leave here the part of DDF about temperature sensor:
```json
{
  "type": "$TYPE_TEMPERATURE_SENSOR",
  "restapi": "/sensors",
  "uuid": [
    "$address.ext",
    "0x01",
    "0x0402"
  ],
  "items": [
    {
      "name": "attr/id"
    },
    {
      "name": "attr/lastannounced"
    },
    {
      "name": "attr/lastseen"
    },
    {
      "name": "attr/manufacturername"
    },
    {
      "name": "attr/modelid"
    },
    {
      "name": "attr/name"
    },
    {
      "name": "attr/swversion"
    },
    {
      "name": "attr/type"
    },
    {
      "name": "attr/uniqueid"
    },
    {
      "name": "config/offset",
      "description": "Relative offset to the main measured value.",
      "default": 0
    },
    {
      "name": "config/on"
    },
    {
      "name": "config/reachable"
    },
    {
      "name": "state/lastupdated"
    },
    {
      "name": "state/temperature",
      "refresh.interval": 60,
      "read": {
        "at": "0x0000",
        "cl": "0x0002",
        "fn": "zcl:attr"
      },
      "write": {
        "fn": "None"
      },
      "parse": {
        "at": "0x0000",
        "cl": "0x0002",
        "eval": "Item.val = (Attr.val * 100) + R.item('config/offset').val",
        "fn": "zcl:attr"
      },
      "default": 0
    }
  ]
}
```

Thanks! Regards!

Smanar, 2024-08-22:
I m still not seing the DDF for the xiaomi_zncz04lm_smart_plug ^^.
Do you want to make a PR to correct something or add something ? Or want I make it ? (to add it officialy)

aramcap, 2024-08-23:
> Hello, I m still not seing the DDF for the xiaomi_zncz04lm_smart_plug ^^. Do you want to make a PR to correct something or add something ? Or want I make it ? (to add it officialy)

Yes! I can do a PR with the changes: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7899