dresden-elektronik/deconz-rest-plugin #7419: Device request for _TZ3000_0zfrhq4i

michaelblight, 2023-12-07:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

SIXWGH

### Manufacturer

_TZ3000_0zfrhq4i

### Model identifier

TS011F

### Device type to add

Switch

### Node info

Note that the switch functionality works indeConz, just not the powere metering.

[image]

[image]

Simple Metering Cluster:
[image]

Electrical Measurement Cluster:
[image]

### Endpoints and clusters

Simple Metering Cluster:
[image]

Electrical Measurement Cluster:
[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering
[image]

### Electrical Measurement
[image]

### Any other cluster of relevance/interest

closed 2024-01-05 by github-actions[bot]

GitTheFOuttaHere, 2024-01-11:
Same problem here. No solution yet.

Smanar, 2024-01-11:
Hello, can try this DDF
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_0zfrhq4i",
  "modelid": "TS011F",
  "vendor": "Tuya",
  "product": "Wired Switch 1 gangs",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 360,
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 1,
            "eval": "Item.val = Attr.val * 10;"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 360
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 360
        },
        {
          "name": "state/voltage",
          "refresh.interval": 360
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 10,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 10,
          "max": 300,
          "change": "0x00000005"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 10,
          "max": 300,
          "change": "0x00000100"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 10,
          "max": 300,
          "change": "0x00000010"
        }
      ]
    }
  ]
}
```

Power/consumption and ect can need to be converted *10 ot /10 by exemple. Just tell me the value you have and the one you need.

michaelblight, 2024-01-12:
Is it possible to use a DDF if I'm using the Home Assistant deCONZ addon? I can connect to the docker container if necessary to test, but I'm not sure what is required.

michaelblight, 2024-01-13:
I tried the above, but after reloading deCONZ it didn't work. When editing the DDF through the GUI it said "unnamed file" (or something like that) and the log said DDF not found.

However, in the meantime I bought another plug from the same supplier, and this one showed up as "_TZ3000_typdpbpg", and was recognised properly. I took a punt and copied the DDF from that one (in /usr/share/deCONZ/devices/tuya/_TZ3000_typdpbpg_smart_plug_eu.json) and renamed the manufacturer name, and that seems to have worked.

Also, so I don't forget next time, I think you need to also reload the deCONZ integration in HA (not just restart the addon), although I have "Allow automatic addition of new devices" disabled, so maybe that's why.

Thanks for your help.

Smanar, 2024-01-13:
But like this, the procedure you have done will be not persistent, it mean at next restart you will loose your changes, it's like that works dockers.

michaelblight, 2024-01-14:
I copied from the source from /usr... into a modified file in /data, which is mapped to a folder on the host, so it's ok.