dresden-elektronik/deconz-rest-plugin #7425: Why is bug devices keep dropping ignored here?

easybeat, 2023-12-10:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

As in another handful bugs described and then silently closed here:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2710
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6201

I'm facing the same problem since I'm using Deconz

[image]
5 Mins. after restart

[image]
15 Mins. after restart

As @dawiinci mentioned here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6201 is this a bug in the firmware?

Why do all the bug reports about this not been followed up?

### Steps to reproduce the behavior

1. start deconz-gui
2. wait 5 Mins. and check connections
3. wait 15 Mins. and check connections again

### Expected behavior

No connections should be dropped

### Environment

- Host system:NAS with Proxmox 8.1
- Running method:  Ubuntu Desktop 20.04 as a Proxmox VM
- Firmware version: 26780700
- deCONZ version: 2.22.00 / 19.9.2022
- Device: ConBee II 
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? If so: USB3 ext. powered Hub + ZWave Stick

easybeat, 2023-12-10:
Here my log files
[deconz log droping lights.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13629114/deconz.log.droping.lights.txt)

Mimiix, 2023-12-10:
The discussion about closing issues is not going to be held here. If you want to have that, open a post on the forums.

For the latter, others are more suitable to answer it.

manup, 2023-12-11:
Does this happen with the latest release too?
There were quite a bunch of fixes in regards to poll and keep devices alive.

easybeat, 2023-12-11:
@manup And just to make clear I have 2 different systems which I have the same symptoms...just one gone completely by now. And this was running since about 3 years but always had this symptoms but whenever I power cycled the lights they came back on. But not now! Powercycle doesn't help.

I have a backup back from June which I try to restore now because there everything was ok.

manup, 2023-12-11:
Currently I wouldn't recommend to switch between versions, but get to the cause where it fails.
Can you please check if DDFs are enabled via *Menu -> Panels -> Control*, should look like:

[image]

The logs above don't show hard networks errors as most/all commands get confirmed but there also appears to be mostly legacy traffic.

With DEV and DDF logging enabled in *Help -> Debug view* we should see more.
It would be helpful to then pick one device which goes unreachable and check what happens there.

Mimiix, 2023-12-11:
> Yes DDF is on ![image](https://private-user-images.githubusercontent.com/37217887/289542930-a9e4cf89-7878-4b99-aef3-353d86ff97ff.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIzMDAyNjIsIm5iZiI6MTcwMjI5OTk2MiwicGF0aCI6Ii8zNzIxNzg4Ny8yODk1NDI5MzAtYTllNGNmODktNzg3OC00Yjk5LWFlZjMtMzUzZDg2ZmY5N2ZmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjExVDEzMDYwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA1M2Y3N2UyOGVmOTJiMWE2YTI3ODU4YWU1N2NmZjcxOTkzNjQ5NmI0YjQyM2IzN2RjYzI2YjUyODIwZmRiZGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.klZr4HQ3wMji5h1xWB5IHn6V-pavF9gjlhhWSZ184lI)
> 
> [DDF error log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13635680/DDF.error.log.txt) here the ddf log

Are you sure the DEV log is enabled on that log ?

easybeat, 2023-12-11:
sorry missed dev
[DDF error log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13635774/DDF.error.log.txt)

manup, 2023-12-11:
>14:08:44:311 DEV Poll Busy /devices/0x00178801011445B8 ZCL response seq: 59, status: 0x86, cluster: 0x0008

Now this looks interesting, which modelid does the 0x00178801011445B8 have – can be seen in the Node Info Panel  (it's a Philips / Signify device)?
The 0x86 means Unsupported Attribute, perhaps the DDF tries to read attributes which are only available in newer lights firmware, have to check with the other devs.

manup, 2023-12-11:
And this Hue Go does drop off after a while as well?
When you power-cycle a light which has dropped off does it come back and can be controlled?

easybeat, 2023-12-11:
No the Hue GO always works it's the lights WAND
[image]

or now even some IKEA Range Extenders
[image]

all different manufactorers failing now

manup, 2023-12-11:
Also can you please provide the firmware version for the Hue Go as shown in the Basic Cluster:
When you double click on the node and then double click on Basic (first item in the list)

Then press "read" in the attributes table:

[image]

manup, 2023-12-11:
> No the Hue GO always works it's the lights WAND [...]

Hmm can't see these in the logs (perhaps too short), but this is still the old deCONZ version? It would be really helpful to make the tests with the recent version.

easybeat, 2023-12-11:
[image]

easybeat, 2023-12-11:
> Hmm can't see these in the logs (perhaps too short), but this is still the old deCONZ version? It would be really helpful to make the tests with the recent version.

well I can do the update to 2.24.2 if you want me too

manup, 2023-12-11:
Ideally DDF + DEV APS, ZDP and INFO over 10-20 minutes. That's quite noisy but should show more what's going on

easybeat, 2023-12-11:
[image]
on 2.24.2 looks the same after a few minutes they seem to connect

[image]
but if I go into Phoscon the lights show all greyed out

easybeat, 2023-12-11:
[Deconz log 1504-1507.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13636500/Deconz.log.1504-1507.txt)
first 3 minutes

easybeat, 2023-12-11:
[Deconz log 1504-1510.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13636548/Deconz.log.1504-1510.txt)
somehow it seems to work

easybeat, 2023-12-11:
[Deconz log 1507-1513.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13636580/Deconz.log.1507-1513.txt)
6 Mins. steps I can do...hope that's ok

easybeat, 2023-12-11:
[Deconz log 1513-1519.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13636664/Deconz.log.1513-1519.txt)
another 6 Mins.

easybeat, 2023-12-11:
[image]
and by the way the Ikea Range Extenders have not been recognized anymore

easybeat, 2023-12-11:
[Deconz log 1519-1525.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13636740/Deconz.log.1519-1525.txt)
that's 21 Mins. do you need more?

manup, 2023-12-11:
Hmm interesting I'm looking after the 0x00158D0000B1EF40 (Wand04 ?)
This one is seen by many devices but no attempt yet to contact it (.. Deconz.log.1507-1513.txt).

This one doesn't have a DDF yet nor the Paul Neuhaus lights, so there seems to be something off with the legacy code :thinking: 

The Ikea repeaters however should work once they are queried (this may take a bit the first time they are used via DDF).

easybeat, 2023-12-11:
[Deconz log 1525-1531.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13636826/Deconz.log.1525-1531.txt)
I think in this 6 minutes light wand 03 lost the green line
[image]

easybeat, 2023-12-11:
> Hmm interesting I'm looking after the 0x00158D0000B1EF40 (Wand04 ?) This one is seen by many devices but no attempt yet to contact it (.. Deconz.log.1507-1513.txt).
> 
> This one doesn't have a DDF yet nor the Paul Neuhaus lights, so there seems to be something off with the legacy code 🤔
> 
> The Ikea repeaters however should work once they are queried (this may take a bit the first time they are used via DDF).

Any more ideas how to get this system back to work...my friends are not very happy that they can't operate their lights :-(

easybeat, 2023-12-11:
[image]
also the Ikea Range Extender are still not recognized

easybeat, 2023-12-11:
@manup Any more help today?

easybeat, 2023-12-11:
Thanks Manu.

yes live session tomorrow would be helpful. I'm available from 08.00 - 11.00 GMT+1

I expect you are in Germany? I'm from Switzerland...so we can then also change to German...

I've sent the backup files to you support e-mail.

easybeat, 2023-12-12:
Thank you for the new version. But how do I have to install this?

I run it on an Ubuntu 20.04 desktop and get this error message:
[image]

easybeat, 2023-12-12:
> Unfortunately I can't manage to be available before 14/15.00.

And yes I could also do these times...let me know

manup, 2023-12-12:
>Thank you for the new version. But how do I have to install this?
>I run it on an Ubuntu 20.04 desktop and get this error message:

Oh damn my bad, thought you're running on a Raspberry Pi. Here is the version for Desktop AMD64

[deconz_2.24.2-debian_amd64.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13648738/deconz_2.24.2-debian_amd64.zip)

easybeat, 2023-12-12:
np...I'll try to install now

easybeat, 2023-12-12:
[image]
It says now that a never version is already installed...should I try it on the machine with the 2.22.0 version?

manup, 2023-12-12:
I guess that's the Ubuntu UI installer thing..
Better make this on the command line:

`sudo dpkg -i deconz_2.24.2-debian_amd64.deb`

easybeat, 2023-12-12:
ok installed it from command line over the 2.22.0 installation...can also try on the box with 2.24.2 already installed

It looks not much different to me
[image]

[image]
in phoscon still all greyed out

here is the log from first 6 Mins.
[Deconz log 20231212 1455-1501.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13648989/Deconz.log.20231212.1455-1501.txt)

easybeat, 2023-12-12:
[Deconz log 20231212 1501-1505.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13649036/Deconz.log.20231212.1501-1505.txt)
next 5 Mins.

manup, 2023-12-12:
Cool thanks, the Ikea range extender and Greeble lights are loaded via DDF now (for Ikea repeater the actual name is shown which wasn't before so it did pick up the code changes here).
I think next it makes sense to let it run for a while to see if it settles on these devices, I'll check the logs

easybeat, 2023-12-12:
[image]
I don't think it works as it has lost a lot of green lines already to Ikea and Wand-Lights

easybeat, 2023-12-12:
[Deconz log 20231212 1508-1511.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13649099/Deconz.log.20231212.1508-1511.txt)

easybeat, 2023-12-12:
[Uploading Deconz log 20231212 1512-1515.txt…]()

manup, 2023-12-12:
Are you also on our deCONZ Discord? Or access via Teamviewer, there I can test a few things.
The logs so far don't show any errors but it seems the devices are not yet treated as reachable

easybeat, 2023-12-12:
ready for tw session....detail on support email

easybeat, 2023-12-12:
np...I'm ready

manup, 2023-12-12:
WTF Teamviewer says session closed for a few minutes

easybeat, 2023-12-12:
come back now

easybeat, 2023-12-12:
come back in

manup, 2023-12-12:
Only 5 minute sessions with a small delay :see_no_evil:  be back in a minute :)

easybeat, 2023-12-12:
[image]
Hmmm at least in Phoscon the ikea's are back!!

manup, 2023-12-12:
Hmmm don't think they are really back, Teamviewer drives crazy, perhaps we can try Anydesk? (see mail).
I think to know why the devices don't react

manup, 2023-12-12:
The network is super weird, I've added a few notes in the text editor, would be good to try the power cycle next to see if some devices manage to come back.

easybeat, 2023-12-12:
Ok. Thanks for you research.

Do you know the "other" network key? Just wondering if there has been a mixup with my other system... 

Is there no way to "migrate" the wrong devices to the new network key?

Powercycle some other devices will only happen tomorrow.

manup, 2023-12-12:
Changing the network key remotely isn't possible yet (there is a Zigbee command to do this but nobody actually uses that, so I assume it's also not really tested/supported for devices).

If you can send me the network key from the other setup I can enter it in the sniffer log and see if it works.

easybeat, 2023-12-12:
And just another thought. Looking at the connected devices now. It looks that even more are lost.
[image]
the 2 hue go have gone now too?

manup, 2023-12-12:
It's due the channel change, they'll be back after switching back or doing a power cycle. It's a bit messy but the Philips devices usually should be the easiest to deal with.

easybeat, 2023-12-12:
Ok. 

Can you find out which is the other network key?

Sent you the key from other system in email

manup, 2023-12-12:
Just checked the key from the mail, it doesn't match any traffic I've captured on channel 15 (ca. 3000 Zigbee messages).
It's not really possible to extract the encryption key from the traffic, only when a network is opened during pairing a key can be sniffed.

easybeat, 2023-12-12:
Hmmm that worries me even more. Where should this network key coming from?

Is there a way in deconz to rgenerate a new network key?

easybeat, 2023-12-13:
@manup Hi Manuel

nothing did work with powercycle and also no more lights were happy to join the network on new channel

[image]

Do you have anymore ideas how to fix this or at least what could have caused it?

manup, 2023-12-13:
Hm damn, but was worth a try often the power cycle can fix issues, but here it seems to broken.
Unfortunately I don't  have more ideas how to bring back the not responding devices other than to factory reset them.

I'd propose to move the network back to channel 15 via Phoscon App > Menu > SGateway Advanced Settings.
Important, **do not** delete the devices in the Phoscon App, so the settings groups and so on don't get lost.

Then start search for lights and reset a device, e.g. for Ikea repeaters this is done via the pin-hole, I think Greeble and Paul Neuhaus do have a reset sequence. (I'd start with the repeaters to see that it works).

----- 

In terms what could have caused this I'm still not totally sure, but the sniffer logs show that there was some nasty stuff with the network keys. Perhaps due a security counter overflow, here it could be that since the `config.ini` wasn't writeable did speed up the problem and deCONZ couldn't track the counter. (I guess at some point deCONZ was started as root user which changed permissions to the file... I fixed the file permissions yesterday).

I'm currently reading through the Zigbee spec to get a better picture how the overflow is usually treated, hopefully to be able to recreate the situation and check what happens via sniffer, but this won't help for your setup since it's already in this unfortunate state.

easybeat, 2023-12-13:
First thank you very much Manuel for all the work you have done and tried to fix it. I really appreciate this!

As this system is running in a friends house I will have a discussion with him if he wants to stay on deconz or not. They will be without light controll in their house for a whole week. I would never have thought this can happen and have all different backup options in place but not for this. I might consider to move everything to zigbee2Mqtt and as there is a ioBroker running to the zigbee adapter from ioBroker. This will mean that operation the Groups and Scenes will be much more difficult as this is for me the biggest feature with deconz/phoscon.

My much bigger worry is that this could also happen on my own system because this is about 3x the size and will be even more stress!

What I don't understand is how this security counter works? Because what happens if I decide to restore a backup from 3 month ago on a complete new environment? Is the counter inside the backup? But then it will be reset also to a much lower number or why do you think there could have been an overflow of the counter?

And next I'm wondering if you maybe should rethink if the config.ini is the right place for the counter to be written? As you describe to start it as deconz as root can happen very easy and if it ends up in such a disaster then there is something wrong by design.

But I let you do your research and would be very interested to know what the problem is. If it helps I could provide you with a backup of the VM from proxmox so you could maybe run your test on this system even if you don't have the lamps?

[21 more replies]

closed 2024-02-11 by github-actions[bot]