dresden-elektronik/deconz-rest-plugin #7438: Aqara E1 QBKG39LM Wall switch no neutral

Kipyneter, 2023-12-11:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Aqara E1 QBKG39LM

### Manufacturer

LUMI

### Model identifier

lumi.switch.b2lc04

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

Clusters:
[image]

[image]

### Basic

[image]

### Further relevant clusters

### On/Off
[image]

### Any other cluster of relevance/interest
Device Temperature Configuration
[image]

LUMI specific:
[image]
[image]
[image]

Multistate input:
[image]

Kipyneter, 2023-12-12:
That did the trick! Thank you so much.
I did not realize that I can change existing files that easy.

SwoopX, 2023-12-12:
Nice. If you could test/note down the values for the Multistate Input Cluster attribute 0x0055 on the button presses, we could create the respective button map as well. Single/double/tripple/long press are the #1 candidates 👍

closed 2024-01-10 by github-actions[bot]