dresden-elektronik/deconz-rest-plugin #7441: ConBee III / 2.24.2

MSL-DA, 2023-12-11:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I can't add any kind of sensor or IKEA blinds (rest-api and Phoscon). Version 2.24.2 shows Conbee instead of Conbee III (docker)

Conbee III is completely useless with this version (2.24.2)

### Steps to reproduce the behavior

Add sensor

### Expected behavior

Nothing happens when I try to add a device on battery

### Environment

- Host system: (Raspberry Pi / PC / NAS / MAC)
- Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
- Firmware version: (26xxyy00)
- deCONZ version (not Home assistant Addon version!): (2.xx.yy)
- Device: (ConBee I / ConBee II / ConBee III / RaspBee I / RaspBee II)
- Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?
- 
Raspberry Pi
Ubuntu/Docker
deCONZ 2.24.2
Conbee III
Yes
No

Mimiix, 2023-12-11:
Can you share some logs while pairing?

Log levels
Aps, aps l2, error, error l2, info info l2.

MSL-DA, 2023-12-11:
I have had to go back to conbee II. I will see if I have time to make a setup with conbee III during the week

Mimiix, 2023-12-11:
Hi

In that case I'm closing this issue. No need to keep it open if we can't debug.

Theres a firmware update for Conbee 3 (as far as I could tell), I suggest testing that and report back to re open this when needed.

closed 2023-12-11 by Mimiix

MSL-DA, 2023-12-11:
You don't think I have updated and tested the latest version? I have. Good for you to close it, it's sufficient for troubleshooting and improvements. I wonder why I bother to keep reporting errors...

Mimiix, 2023-12-11:
> You don't think I have updated and tested the latest version? I have. Good for you to close it, it's sufficient for troubleshooting and improvements. I wonder why I bother to keep reporting errors...

Hey I'm happy to re open when you get back to it.