dresden-elektronik/deconz-rest-plugin #7448: Device wrongfully recognized (Human Presence Detector, _TZE200_v6ossqfy)

raderack3948, 2023-12-13:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

The sensor is a MmWave presence detector, I've also found it listed as such in the code here in the repo. But when I connect it trough the phoscon app it is shown as a light switch?
Note: I'm using the deCONZ integration in Home Assistant.

Phoscon Version:
Version: 2.22.02 / 19/09/2022
Firmware: 26720700

Phoscon View:
[image]

Deconz View:
[image]

Node Info:
[image]

Basic Info:
[image]

### Steps to reproduce the behavior

Occurs on binding the sensor

### Expected behavior

That the device shows up as a motion sensor

### Environment

- Host system: Home Assistant
- Running method: HASS OS
- Firmware version: 26720700
- deCONZ version (not Home assistant Addon version!): 2.22.02
- Device: ConBee II
- Do you use an USB extension cable: no
- Is there any other USB or serial devices connected to the host system?: No

Mimiix, 2023-12-13:
Hi,

This please read the template carefully. You are not reporting an issue with deCONZ but Phoscon.

Please use the correct channels.

Closing this.

Kind regards,

closed 2023-12-13 by Mimiix

raderack3948, 2023-12-13:
Sorry for asking - where do I find the phoscon repo? The phoscon repo here is archived, I cannot create an issue.

Mimiix, 2023-12-13:
> Sorry for asking - where do I find the phoscon repo? The phoscon repo here is archived, I cannot create an issue.

Thats shown during creation of this issue. There a url to the forums.

raderack3948, 2023-12-13:
> > Sorry for asking - where do I find the phoscon repo? The phoscon repo here is archived, I cannot create an issue.
> 
> Thats shown during creation of this issue. There a url to the forums.

Found it - Thank you!