dresden-elektronik/deconz-rest-plugin #7450: IKEA SOMRIG Shortcut Button support

cook2000, 2023-12-13:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

IKEA SOMRIG Shortcut Button

### Manufacturer

IKEA of Sweden

### Model identifier

SOMRIG shortcut button

### Device type to add

None

### Node info

[image]

### Endpoints and clusters

[image]

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### Poll Control
[image]

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest
[image]

Mimiix, 2023-12-13:
Can you please add all screenshots?

cook2000, 2023-12-13:
added some more screenshots above and updated.

Idaho947, 2023-12-27:
I bought this sensor and can t connect it to deconz. @Smanar have you a magic ddf ?

Smanar, 2023-12-27:
Can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "IKEA of Sweden",
  "modelid": "SOMRIG shortcut button",
  "product": "SOMRIG Shortcut Button",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xFC80"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 86400,
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3200,
          "max": 3600,
          "change": "0x00000002"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0xFC80"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0xFC80"
    }
  ]
}
```

It will enable and configure the device (Need to use the 0xFC80 cluster) but it will not work yet.
After that (like usual), just enable deconz log with "info" + "info_l2" and share error about missing button map.

Idaho947, 2023-12-28:
The device is paring but I see nothing in log when I press the button

Smanar, 2023-12-29:
```
        "ikeaShortcutMap": {
            "vendor": "IKEA of Sweden",
            "doc": "IKEA SOMRIG Shortcut Button",
            "modelids": ["SOMRIG Shortcut Button"],
            "map": [
                [1, "0x01", "FC80", "COMMAND_3", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Button 1"],
                [1, "0x02", "FC80", "COMMAND_6", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Button 2"]
            ]
        },
```

But I think you can do some optimisation with this code from Z2M

```
            const lookup = {
                commandAction1: 'initial_press',
                commandAction2: 'long_press',
                commandAction3: 'short_release',
                commandAction4: 'long_release',
                commandAction6: 'double_press',
            };
```

On deconz value to use are

```
        "FC80": {
            "COMMAND_1": 1,
            "COMMAND_2": 2,
            "COMMAND_3": 3,
            "COMMAND_4": 4,
            "COMMAND_6": 6
        },
```

Smanar, 2023-12-29:
For exemple by deduction to have a double press on button 1

`[1, "0x01", "FC80", "COMMAND_6", "0", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "Button 2 double press"]`

For a long press on button 2

`[1, "0x02", "FC80", "COMMAND_2", "0", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Button 2 long press"]`

Idaho947, 2023-12-29:
What do you think of :

`"ikeaSomrigMap": {
            "vendor": "IKEA of Sweden",
            "doc": "IKEA SOMRIG Shortcut Button",
            "modelids": ["SOMRIG Shortcut Button"],
            "map": [
                [1, "0x01", "FC80", "COMMAND_3", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Button 1"],
				[1, "0x01", "FC80", "COMMAND_4", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Button 1"],
				[1, "0x01", "FC80", "COMMAND_6", "0", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "Button 1"],
                [1, "0x02", "FC80", "COMMAND_3", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Button 2"],
				[1, "0x02", "FC80", "COMMAND_4", "0", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Button 2"]
				[1, "0x02", "FC80", "COMMAND_6", "0", "S_BUTTON_2", "S_BUTTON_ACTION_DOUBLE_PRESS", "Button 2"]
            ]
        },
`

Smanar, 2023-12-29:
You have the device ^^, you can test your work.

On deconz we don't use the S_BUTTON_ACTION_INITIAL_PRESS, useless, we use instead the S_BUTTON_ACTION_SHORT_RELEASED to trigger event when the button is released.

Value can be different from z2m, but easy to correct whith try, else it seem good for me.

Idaho947, 2023-12-29:
Ok it doen't work :

When I press on spot one time :
[image]

When I press two spot one time :
[image]

My buttom map :
[image]

Idaho947, 2023-12-29:
yes that's why :

[image]

For me it's working.

SwoopX, 2023-12-29:
@Idaho947 If you could share in text form what's working for you, we could quickly add it to the upcoming release 😉

Idaho947, 2023-12-30:
I made that. Hope it's Ok :
`    "ikeaSomrigMap": {
            "vendor": "IKEA of Sweden",
            "doc": "IKEA SOMRIG Shortcut Button",
            "modelids": ["SOMRIG shortcut button"],
            "map": [
                [1, "0x01", "FC80", "COMMAND_2", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Button 1"],
                [1, "0x01", "FC80", "COMMAND_3", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Button 1"],
                [1, "0x01", "FC80", "COMMAND_4", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Button 1"],
                [1, "0x01", "FC80", "COMMAND_6", "0", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "Button 1"],
                [1, "0x02", "FC80", "COMMAND_2", "0", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Button 2"],
                [1, "0x02", "FC80", "COMMAND_3", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Button 2"],
                [1, "0x02", "FC80", "COMMAND_4", "0", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Button 2"],
                [1, "0x02", "FC80", "COMMAND_6", "0", "S_BUTTON_2", "S_BUTTON_ACTION_DOUBLE_PRESS", "Button 2"]
            ]
        }, `

Smanar, 2023-12-30:
Ha right, I have made a typo ^^.
So thx for the corrected/tested button map :).

One of you want to make the PR ?

noud-github, 2024-01-03:
I made a DDF file and changed buttons file here:

https://github.com/noud-github/deconz-rest-plugin/tree/somrig

not sure what the rules are for a PR, need to read up on that first ;-)

Smanar, 2024-01-03:
There is an online validator for "rules" now.
Else there is a problem, you have created a buttonmap file in the "device" folder, you just need to edit the existing one (on the root)

noud-github, 2024-01-03:
after testing I moved it back to the wrong folder, fixed now, 

simple question does the validator run on the |PR if I creat it?

CerealPT, 2024-01-04:
@noud-github 
I updated my button_maps.json in the folder /usr/share/deCONZ/devices and also added the new DDF to the /devices/ikea.

After that I restarted deCONZ, paired the somrig shortcut button and when I press the buttons I see the following in the logs:

No button map for: SOMRIG shortcut button, unicast to 0x0000, endpoint 0x01, custer: FC80 (0xFC80), command: COMMAND_1 (0x01), payload: 00, zclseq: 95

What can be a cause for this? IS the buttons_map.json correctly loaded? is there something wrong there?

Thanks for your support!

noud-github, 2024-01-04:
I added the buttons file in the wrong folder it should be : `/usr/share/deCONZ/`.  not /usr/share/deCONZ/devices

noud-github, 2024-01-04:
> I noticed that when I restart the deCONZ addon in Homeassistant, the button_maps.json file inside the deCONZ docker container is overwritten and the mappings are lost. So I might not be updating the correct file.

That is docker behavior, I use command `docker compose restart` if you know hte conatainer name:
(docker ps) than you cloud do `docker restart <containername>`  from a CLI

Smanar, 2024-01-04:
>simple question does the validator run on the |PR if I creat it?

> All checks have passed
> 2 successful checks
> @github-actions
> Verify DDF JSON / validate (pull_request) Successful in 6s
> [Details](https://github.com/dresden-elektronik/deconz-rest-plugin/actions/runs/7401026445/job/20166057464?pr=7511)
> @github-actions
> Verify button maps / build (pull_request) Successful in 12s
> [Details](https://github.com/dresden-elektronik/deconz-rest-plugin/actions/runs/7401026458/job/20166057474?pr=7511)

All is fine for me ^^

CerealPT, 2024-01-04:
> > I noticed that when I restart the deCONZ addon in Homeassistant, the button_maps.json file inside the deCONZ docker container is overwritten and the mappings are lost. So I might not be updating the correct file.
> 
> defualt `docker compose DOWN` will remove the container inc your edits
> 
> That is docker behavior, I use command `docker compose restart` if you know hte conatainer name: (docker ps) than you cloud do `docker restart <containername>` from a CLI
> 
> other option is to map the changed files into your container so the will be there (mount -v option on docker)

Hi @noud-github!

Thanks for the hint!

I am now able to see the events inside Homeassistant, nevertheless I can't pick the right event id, in order to create an automation.

Is that something that needs to be done in the DDF?

noud-github, 2024-01-04:
@CerealPT  I used this blueprint: https://github.com/noud-github/ha-blueprints

CerealPT, 2024-01-04:
> @CerealPT I used this blueprint: https://github.com/noud-github/ha-blueprints
> 
> does a nice mapping between events and button names you can understand ;-)

Man....this is so sweet 🤟 
Made my life really easier!

Thanks a lot and hope to see all of these new developments on the new deconz release!

ebaauw, 2024-01-05:
Picked one up today.  My observations:
- Indeed, it only reports button events through the 0xFC80 cluster.  Endpoint 0x01 is for the upper button (one dot); endpoint 0x02 for the lower button (two dots);  Binding the _On/Off_ or _Level Control_ clusters seems to have no effect?  From what I can tell, it sends the following commands from 0xFC80, which should be mapped to:
  Command | Action | Buttonevent
  -- | -- | --
  0x01 | Initial Press | x000
  0x02 | Hold | x001
  0x03 | Release (after press) | x002
  0x04 | Release (after hold) | x003
  0x06 | Double press | x004
- It does react to _Identify_, so should expose `config/alert`.
- The 0xFC80 bindings should go first; the device goes incommunicado without such binding.  Even pressing the link button won't wake it, but removing and re-inserting the battery will;
- It should expose `attr/productid`, after _Product Code_;
- _Image Type_ is 0x3B08.
- Maybe some setting on the (server) 0xFC80 cluster changes the behaviour for direct bindings of the (client) _On/Off_ and _Level Control_ clusters.  Need to check with the Dirigera hub.

Smanar, 2024-01-05:
For supported command it use

```
            "map": [
                [1, "0x01", "FC80", "COMMAND_1", "0", "S_BUTTON_1", "S_BUTTON_ACTION_INITIAL_PRESS", "0x01"],
                [1, "0x01", "FC80", "COMMAND_2", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "0x02"],
                [1, "0x01", "FC80", "COMMAND_3", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "0x03"],
                [1, "0x01", "FC80", "COMMAND_4", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "0x04"],
                [1, "0x01", "FC80", "COMMAND_6", "0", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "0x06"],
                [1, "0x02", "FC80", "COMMAND_1", "0", "S_BUTTON_2", "S_BUTTON_ACTION_INITIAL_PRESS", "0x01"],
                [1, "0x02", "FC80", "COMMAND_2", "0", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "0x02"],
                [1, "0x02", "FC80", "COMMAND_3", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "0x03"],
                [1, "0x02", "FC80", "COMMAND_4", "0", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "0x04"],
                [1, "0x02", "FC80", "COMMAND_6", "0", "S_BUTTON_2", "S_BUTTON_ACTION_DOUBLE_PRESS", "0x06"]
            ]
        },
```

So all is fine, on this side.

closed 2024-01-22 by manup

dklinger, 2024-03-06:
Can there be support added for the battery of the SOMRIG? It's currently showing 0%.

Smanar, 2024-03-06:
You have just included it ? Can need time, because I m looking the DDF again, I don't see something bad.

dklinger, 2024-03-07:
@Smanar Yes, I included it yesterday. How long can/should it take for the correct value to show up?

Smanar, 2024-03-07:
Ha have see something, there is no poll in case the bind is not working
Can try to replace

```
        {
          "name": "config/battery",
          "refresh.interval": 86400,
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          }
        },
```

By

```
        {
          "name": "config/battery",
          "refresh.interval": 86400,
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          },
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          }
        },
```

After 24h if the bind is not working deconz will poll itself the device to have the value.

DeviDarkL, 2024-04-01:
Hello, thank you for doing this integration. Battery still shows 0%. Will there be any fix?

Smanar, 2024-04-01:
With the changes on previous post it still doesn't work ?

DeviDarkL, 2024-04-01:
Hey, I am sorry but I am a total noob on this. I only added to deconz but neither the battery nor the button mapping works after adding the device. What DDF file are you using for this remote? Does someone has it with all the modifications from this conversation?

Best regards

Smanar, 2024-04-01:
Make a try with this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/ikea/somrig_shortcut_button.json
And this changes https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7450#issuecomment-1983908694