dresden-elektronik/deconz-rest-plugin #7453: Cannot install Deconz 2.24.2 on DietPi v8.24 as VM on Proxmox

stoneubi, 2023-12-14:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Hi there, 

for many years I used a deconz on a dietpi vm no promox without any problems. 

Since the update to dietpi 8.24 and the update to the latest deconz version 2.24.2 the system isn't working any more. 

Therefore I started from scratch, created to new VM, updated dietpi to the latest version and tried to install deconz again with GUI according to this guide: https://phoscon.de/de/conbee2/install#raspbian

During the installation nothing seems to be wrong but the menu entry programming and deconz is missing. Also there is no /usr/bin/deCONZ-autostart.sh file and I'm not able to start the GUI version or even the headless version. 

Is there any known issue regarding this?

I tried the docker version and this works fine on my dietpi vm. 

### Steps to reproduce the behavior

1. Create a new dietpi VM in proxmox
2. add the usb conbee II stick to this vm
3. start dietpi and make your settings to start in desktop version
4. login to the desktop, open terminal and try to install deconz 
5. Check the result and recognize that deconz isn't working

### Expected behavior

Deconz latest version should work an vm dietpi on proxmox as all the years before. 

### Environment

- Host system: promox 8.1.3
- Running method: diepti VM 
- Firmware version: 26720700
- deCONZ version: 2.24.2
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system?: no

stoneubi, 2024-01-05:
Is there any bugfix planned?

Mimiix, 2024-01-05:
This one slipped me somehow. Sorry for the late response.

Your trying to install on a OS that isn't supported as of the list.  Nevertheless, this issue is not as described in #5113. 

I recommend opening a post on the forums. 

Closing.

closed 2024-01-05 by Mimiix