dresden-elektronik/deconz-rest-plugin #7454: IKEA Vallhorn motion sensor device request

MagnusKazen, 2023-12-14:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

IKEA Vallhorn wireless motion sensor

### Manufacturer

IKEA of Sweden

### Model identifier

Vallhorn wireless motion sensor

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off
[image]
Just blank, seems to be no info here.
### Level Control
None
### Color Control
None
### Thermostat
None
### Simple Metering
None
### Electrical Measurement
None
### Any other cluster of relevance/interest
[image]
[image]
[image]
[image]
[image]

Just let me know if there is anything more You need. I'm pretty new at this...

hyper2910, 2024-01-02:
any idea when Vallhorn is implemented?

noud-github, 2024-01-03:
I made a DDF file and js file here

https://github.com/noud-github/deconz-rest-plugin/tree/vallhorn

not sure what the rules are for a PR, need to read up on that first ;-)

Mimiix, 2024-01-04:
@noud-github  If you do a PR, theres a check automation running :)

@Smanar Perhaps you can help :)?

Smanar, 2024-01-04:
I m answering on the PR page.

Mimiix, 2024-01-18:
> any update?

Did you read the PR?

It has been merged. Support should be added in upcommimg 2.25.2.

closed 2024-01-18 by Mimiix

ebaauw, 2024-01-22:
(duplicate from my comment on the PR)

Finally managed to get hold of one of these.  As suggested by the screenshots, it's now a proper motion sensor on endpoint 0x02.  It sends "no more motion detected" events, and honours _PIR Occupied to Unoccuplied Delay_, which should be exposed as `config/delay` (cf. the Hue motion sensors).  As far as I can tell, it detects the next motion immediately after the delay has passed (so no cooldown period).

It still sends _On with Timed Off_ from endpoint 0x01, like the Trådfri motion "sensors" (or rather controllers).  It seems to use the configured _On Time_ before resending this command.  Haven't yet checked whether it still supports group bindings; if it does, we should expose `config/group`.

I think we should expose two ZHAPresence resources, one for the controller function on endpoint 0x01 (with `state/daylight`, `config/group`, and read-only `config/delay`, used for deCONZ to reset `state/presence`, cf the Trådfri motion "sensors") and one for the sensor function on 0x02 (with read/write `config/delay`, cf. the Hue motion sensors).

Daniel-Ebert, 2024-02-06:
Hi there, I just also put mine into service and noticed, that the "Lux" information is always 0 in my case. Is this a known issue, or does the light sensor work for anybody?
I have version 2.25.2-beta installed (the first one adding "built in" support for Vallhorn)

Best regards

Daniel

Daniel-Ebert, 2024-02-06:
Hi there @hyper2910 , thanks for your quick response! Did you also use the DDF delivered in the newest beta release, or did you use something from this thread here? And are you using the "official" LADDA batteries/ Batteries with 1,2V also specified in the manual? I've read in the Home Assistant forum, that Batteries with 1,5V do make VALLHORN behave weird.
(I currently have 1,5V ones installed)

hyper2910, 2024-02-06:
Hi,

Using the ddfs from the thread, also using Lada Batteries. Still at 2.25.1

marmic-hub, 2024-02-07:
They don't work for me either. I used the ddf from this thread.
[image]

Daniel-Ebert, 2024-02-07:
> Hi,
> 
> Using the ddfs from the thread, also using Lada Batteries. Still at 2.25.1

Okay, thanks for the information! I just bought 1,2V accumulators ( replacing the batteries with 1,5V I have put in initially) and I will update you if it works with them!
In these threads they have discussed it quite actively, that the Vallhorns seem to have problems with 1,5V batteries:
https://community.home-assistant.io/t/ikea-vallhorn-false-motion-detection/654298/22
https://forum.iobroker.net/topic/71681/gel%C3%B6st-ikea-vallhorn-bewegungsmelder-l%C3%B6st-st%C3%A4ndig-aus/21

Daniel-Ebert, 2024-02-07:
I tested it with the 1,2V accumulators and at least for me the false detections disappeared (the sensor triggered approx. every minute falsefully with 1,5V alkalines).
The Lux sensor though does not work for me yet.

marmic-hub, 2024-02-07:
I have 1.5V alkaline batteries in use and **no** false detections.
However, I do not get a the lux information either. 
Based on other posts, I assume that the sensor is simply not providing this information. 

It would be nice if someone could post a screenshot from deconz of a functioning lux display.

SwoopX, 2024-02-07:
> It would be nice if someone could post a screenshot from deconz of a functioning lux display.

Please see the last screenshot of the first post showing the illuminance value.

hyper2910, 2024-02-08:
[image]
[image]

marmic-hub, 2024-02-08:
Thank you for the hint. I will check whether I need to do the binding again. Maybe that will solve the problem

Daniel-Ebert, 2024-02-08:
Thanks for the provided steps to do this! Just to clarify for me: So the DDF is missing the binding information and in a future version this also will work automatically?

SwoopX, 2024-02-08:
Correct. As part for illuminance measurement is entirely missing, there cannot be any set binding and reporting (except when added manually).

It's rather easy to add and once it is part of the DDF, it will be handles automatically.

marmic-hub, 2024-02-08:
Perfect, it worked for me!
[image]

noud-github, 2024-02-09:
created a new PR with the add items

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7584

Daniel-Ebert, 2024-02-14:
Just as short feedback: Works perfectly with your new DDF-Additions in the latest beta :) 
Thank you very much again!

Best regards

Daniel