dresden-elektronik/deconz-rest-plugin #7455: Request to add support for GLEDOPTO GL-S-004P

rstein5, 2023-12-16:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

GLEDOPTO GL-S-004P

### Manufacturer

GLEDOPTO

### Model identifier

GL-S-004P

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

rstein5, 2023-12-16:
Diese RGBCCT Variante von ZigBee MR16 Spots arbeitet mit ZigBee 3.0 und ist so mit den gängigsten Gateways kompatibel.
Dieser Spot ist kann die Farbtemperatur zwischen 2200 und 6500 Kelvin einstellen sowie als RGB Farbwechsler, je nach Steuerung und Gateway auch zeitgleich um die Farbsättigung zu regulieren.

Mimiix, 2023-12-16:
> Diese RGBCCT Variante von ZigBee MR16 Spots arbeitet mit ZigBee 3.0 und ist so mit den gängigsten Gateways kompatibel. Dieser Spot ist kann die Farbtemperatur zwischen 2200 und 6500 Kelvin [...]

Please post in English.

rstein5, 2023-12-16:
I translated the comment and uploaded the missing information. I hope they are complete now. Sorry, but it's my first post here

closed 2024-01-14 by github-actions[bot]