dresden-elektronik/deconz-rest-plugin #7456: battery status

sanluca, 2023-12-16:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Hi everyone,

I am unable to see the battery status of the two-button remotes. I have taken screenshots and attached the ddf file if anyone can help me.

Thanks in advance

`{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "_TZ3000_5e235jpa",
    "_TZ3000_t8hzpgnd",
    "_TZ3000_tzvbimpq",
    "_TZ3000_i3rjdrwu"
  ],
  "modelid": [
    "TS0042",
    "TS0042",
    "TS0042",
    "TS0042"
  ],
  "product": "Tuya remote 2 gangs",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/tuya/_TZ3000_TS0042_2gang_remote.json",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 86400,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}`

### Steps to reproduce the behavior

the problem is always present

### Expected behavior

to see the battery status

### Screenshots

[image]
[image]
[image]
[image]
[image]

### Environment

- Host system: (Raspberry Pi / PC / NAS / MAC)
- Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
- Firmware version: (26xxyy00)
- deCONZ version (not Home assistant Addon version!): (2.xx.yy)
- Device: (ConBee I / ConBee II / ConBee III / RaspBee I / RaspBee II)
- Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?

Smanar, 2023-12-16:
You haven't them direclty on the API too ?
It's a long story but there was 2  kind of device.

The old one need a battery bind, and not the second one else it cause battery drain, so was removed for all of them.

Can try ro add on your DDF

```
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3200,
          "max": 3600,
          "change": "0x00000002"
        }
      ]
    }
```

sanluca, 2023-12-19:
Yes, my remote controls used to drain the battery immediately at first, so I guess I won't leave it without
thanks

Smanar, 2023-12-19:
But on second version, with the battery drain, the report is automatic, you don't need the bind.

You haven't value after 24 ?

Try adding to the DDF

```
        {
          "name": "config/tuya_unlock"
        },
```

And re-include the device.

sanluca, 2024-01-13:
I tried to modify it, but it drained the battery again, I'll leave it without battery control, maybe in the future I'll look for other remote controls, thanks

closed 2024-02-13 by github-actions[bot]