dresden-elektronik/deconz-rest-plugin #7457: LQI display no longer works after V2.24.00

TC1666, 2023-12-16:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Using deconz v2.24.00 on Windows7, LQI values are displayed on the node map.
[image]

Updating to any version after 2.24.00, such as .01 or .02, then LQI values are no longer displayed.

### Steps to reproduce the behavior

1. version v2.24.00 or earlier displays LQI values.
2. update to v2.24.01 or v2.24.01 and LQI values no longer display.
3. Revert back to v2.24.00 and LQI values appear again.

### Expected behavior

LQI values should display, are essential to troubleshooting network issues.

### Environment

- Host system: Intel hardware
- Running method: Windows 7 Enterprise
- Firmware version: 26780700
- deCONZ version (not Home assistant Addon version!): 2.24.xx
- Device: ConBee II
- Do you use an USB extension cable: Yes
- Is there any other USB or serial devices connected to the host system? No

SwoopX, 2023-12-22:
Indeed, also nothing to see under Linux with v2.25.0

TC1666, 2024-01-01:
Hi, is this going to be fixed in the next release? I recently started to have network problems and the first clue in helping me troubleshoot was the LQI readings. I cannot move forward in new deconz updates (and getting better device support) until this feature works again. Thanks!

Mimiix, 2024-01-01:
> Hi, is this going to be fixed in the next release? I recently started to have network problems and the first clue in helping me troubleshoot was the LQI readings. I cannot move forward in new deconz updates (and getting better device support) until this feature works again. Thanks!

For device support you don't need new versions. You can load the DDF manually :)

TC1666, 2024-01-01:
Thanks @Mimiix , but as you know sometimes there are code changes that go beyond DDFs that also help. My question still stands as to whether this function will be fixed soon? Having LQI readings in my opinion provides much value.

Mimiix, 2024-01-01:
> Thanks @Mimiix , but as you know sometimes there are code changes that go beyond DDFs that also help. My question still stands as to whether this function will be fixed soon? Having LQI readings in my opinion provides much value.

I've asked Manup. Hope he'll reply this week.

Don't get me wrong, just wanted to give you some other ways if it was just device requests you were waiting on :)

manup, 2024-01-01:
The LQI values will come back soon, unfortunately didn't make it in v2.25.1 but should be in v2.25.2.
Background: I forgot to implement them after refactoring the UI code to use less CPU for drawing nodes and link lines.

manup, 2024-02-05:
Issue addressed in v2.25.2, drawing the values is still a bit slow but should be ok for debugging purpose.

closed 2024-02-05 by manup