dresden-elektronik/deconz-rest-plugin #7459: Ubisys S1 (5501) and S2 (5502) configuration: Unable to change button mode (could not set attribute)

steinbachd, 2023-12-17:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

It is not possible to configure the button mode for Ubisys S1(5501) and S2 (5502) through the REST API. When issuing the corresponding command, an error message is returned instead. 

The current mode is also missing (null) when querying the sensor configuration. 

Interestingly the command works with an S2-R device and there the current mode is also displayed properly.

### Steps to reproduce the behavior

Issue the following command to change the button mode: 
`curl -X PUT -i http://localhost:8080/api/(api-key)/sensors/(sensor-id)/config --data '{ "mode": "momentary" }'`

Response:
`[{"error":{"address":"/sensors/103/config/mode","description":"Could not set attribute","type":608}}]`

### Expected behavior

I would expect the message `[{"success":{"/sensors/(sensor-id)/config/mode":"momentary"}}]`

This is confirmed working for S2-R

### Environment

- Host system: PC
- Running method: Home Assistent deCONZ Add-on
- Firmware version: 26720700
- deCONZ version (not Home assistant Addon version!): (2.22.2)
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? No

jensflorian, 2023-12-18:
Can confirm this problem, however this is device independent. One recently added S2-R device is shown:
`    "config": {
        "group": "20010,20011",
        "mode": null,
        "on": true,
        "reachable": true
    },
`
An older S2-R device is correctly configured:
`    "config": {
        "group": "20006,20007",
        "mode": "momentary",
        "on": true,
        "reachable": true
    },`
I think that has something to do with the change from legacy code to DDFs.

steinbachd, 2023-12-18:
Thanks for checking. I was briefly screening the code and could not find a specific reference to S2-R that would explain the different behavior. So it might be not related to the model but rather something else.

SwoopX, 2023-12-19:
This is indeed a bug, I'll explain later. The preferred fix should go via DDF modification, but that should be tested. Anybody who could do that?

steinbachd, 2023-12-20:
Would need to set up the dev environment which might take some time. If someone else has this already (and the corresponding device) that might be easier, but happy to give it a go if needed.

SwoopX, 2023-12-20:
So, apparently, the code is stepping on its own foot here. For a DDF to be able to really use `config/mode`, a special meta item needs to be set, which is missing for all ubisys devices offering a ZHASwitch resource. However, the legacy code for setting the mode is still available, but is never reached due to code precedence.

As indicated, it should be feasible to get also that fully covered by DDF. The below excerpt is the **complete** ZHASwitch part of the S2/S2-R DDF, which must replace the current state. Hoping that I haven't made a mistake in the Byte order, it should allow to set rocker and momentary mode.

If the device is e.g. currently set to `rocker`, it must first be set to `momentary` and then back to `rocker` for the configuration to take effect. Once confirmed to be working, I'm more than happy to raise the required PRs for all ubisys devices.

```
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x03",
        "0x0006"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0001",
        "endpoint": "0x03",
        "in": [
          "0x0000",
          "0x0003"
        ],
        "out": [
          "0x0005",
          "0x0006",
          "0x0008"
        ]
      },
      "meta": {
        "values": {
          "config/mode": {"momentary": 0, "rocker": 1, "custom": 2}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/group",
          "default": "auto,auto"
        },
        {
          "name": "config/mode",
          "write": {
            "fn": "zcl:cmd",
            "ep": "0xE8",
            "cl": "0xFC00",
            "cmd": "0x02",
            "eval": "if (Item.val == 'momentary') { '01004841020006000D0306000206010D04060002' } else if (Item.val == 'rocker') { '00014841040006000D0306000206010D040600020600030306000206010304060002' } else if (Item.val == 'custom') { '000148' }"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
```

jensflorian, 2023-12-24:
This is the data stored under E8 - FC00 (Device Setup):
[image]
My S2-R is configured as momentary and this seems to match the default settings shown on page 28 in the Ubisys manual:
https://www.ubisys.de/wp-content/uploads/ubisys-s2-technical-reference.pdf

I however noticed in your code that this is preceded by '010048' for momentary and by '000148' for rocker and custom. Are these values correct and need to be part of the array sent to the device?

jensflorian, 2023-12-26:
With this code, mode can be sucessfully written:
`[
    {
        "success": {
            "/sensors/50/config/mode": "rocker"
        }
    }
]`

However the physical response of the device remains unchanged (in my case still acting as a momentary) and also reading the FC00 comfirms that the stored array remains unchanged. Rebooting the device also didn't help either.

According to this old theread (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1392#issuecomment-478339770) you have a S2. Does your above DDF code work with your S2 and settings for S2-R need to be adapted?

SwoopX, 2023-12-26:
@jensflorian the referenced PR contains working code 😉

closed 2023-12-27 by manup

steinbachd, 2023-12-28:
Thanks a lot for the PR, I have pulled the latest version and can confirm the observation from @jensflorian above. Everything is displayed correctly, but the device itself was not modified. I was trying with an S2 5502.

SwoopX, 2023-12-29:
Well, seems to look fine from my end based on the PR for the S2

```
21:10:19:267 ZCL attribute report 0x001FEE0000000000 for cluster: 0x0006, ep: 0x02, frame control: 0x18, mfcode: 0x0000
21:10:19:267    payload: 00001000
21:10:19:269 ZCL read attr 0x001FEE0000000000, ep: 0xE8, cl: 0xFC00, attr: 0x0001, mfcode: 0x0000, aps.id: 13, zcl.seq: 63
21:10:19:270 SC tick --> StateRead config/mode, 00:1f:ee:00:00:00:00:00-03-0006
21:10:19:372 ZCL got data for node=0x2BD6, cl=0xFC00, at=0x0001, status=0x00, type=0x48
21:10:19:375 expression: var m = ''; for (var i = 4; i < ZclFrame.payloadSize; i++) { m += ('00' + ZclFrame.at(i).toString(16)).slice(-2); } m = m.toUpperCase(); if ( m == '41020006000D0306000206010D04060002') { Item.val = 'momentary'; } else if ( m == '41040006000D0306000206010D040600020600030306000206010304060002') { Item.val = 'rocker'; } else { Item.val = 'custom_' + m; } --> momentary
21:10:19:377 [INFO] - No button handler for: S2 (5502), unicast to: 0x0000, endpoint: 0xE8, cluster: 0xFC00, command: 0x01, payload: 0100004841020006000D0306000206010D04060002, zclSeq: 63
21:10:19:378 SC config/mode: synced
21:10:19:379 SC --> StateFinished
21:10:19:379 SC state change finished: 00:1f:ee:00:00:00:00:00-03-0006

21:12:15:612 Websocket 127.0.0.1:43464 send message: {"config":{"mode":"rocker"},"e":"changed","id":"49","r":"sensors","t":"event","uniqueid":"00:1f:ee:00:00:00:00:00-03-0006"} (ret = 16)
21:12:15:616 SC tick --> StateCallFunction
21:12:15:619 00:1f:ee:00:00:00:00:00-03-0006/config/mode expression: if (Item.val == 'momentary') { '01004841020006000D0306000206010D04060002' } else if (Item.val == 'rocker') { '01004841040006000D0306000206010D040600020600030306000206010304060002' } else if (Item.val.slice(0,7) == 'custom_') { '010048' + Item.val.slice(7, Item.val.length) } --> 01004841040006000D0306000206010D040600020600030306000206010304060002
21:12:15:622 ZCL cmd attr 0x001FEE0000000000, ep: 0xE8, cl: 0xFC00, cmd: 0x02, mfcode: 0x0000, aps.id: 190, zcl.seq: 215
21:12:15:983 [INFO] - No button handler for: S2 (5502), unicast to: 0x0000, endpoint: 0xE8, cluster: 0xFC00, command: 0x04, payload: 00, zclSeq: 215

21:12:24:615 DEV Poll Busy /devices/0x001FEE0000000000 APS-DATA.confirm id: 103, ZCL seq: 245, status: 0x00
21:12:24:642 ZCL got data for node=0x2BD6, cl=0xFC00, at=0x0001, status=0x00, type=0x48
21:12:24:647 expression: var m = ''; for (var i = 4; i < ZclFrame.payloadSize; i++) { m += ('00' + ZclFrame.at(i).toString(16)).slice(-2); } m = m.toUpperCase(); if ( m == '41020006000D0306000206010D04060002') { Item.val = 'momentary'; } else if ( m == '41040006000D0306000206010D040600020600030306000206010304060002') { Item.val = 'rocker'; } else { Item.val = 'custom_' + m; } --> rocker
21:12:24:650 [INFO] - No button handler for: S2 (5502), unicast to: 0x0000, endpoint: 0xE8, cluster: 0xFC00, command: 0x01, payload: 0100004841040006000D0306000206010D040600020600030306000206010304060002, zclSeq: 244
21:12:24:652 SC config/mode: synced
21:12:24:654 SC --> StateFinished
21:12:24:655 SC state change finished: 00:1f:ee:00:00:00:00:00-03-0006
21:12:24:709 ZCL got data for node=0x2BD6, cl=0xFC00, at=0x0001, status=0x00, type=0x48

21:13:09:919 INSERT INTO resource_items (sub_device_id,item,value,source,timestamp) SELECT id, 'config/mode', 'momentary', 'dev', 1703880789 FROM sub_devices WHERE uniqueid = '00:1f:ee:00:00:00:00:00-03-0006'
21:13:09:939 Websocket 127.0.0.1:43454 send message: {"config":{"mode":"momentary"},"e":"changed","id":"49","r":"sensors","t":"event","uniqueid":"00:1f:ee:00:00:00:00:00-03-0006"} (ret = 16)
21:13:09:942 Websocket 127.0.0.1:43464 send message: {"config":{"mode":"momentary"},"e":"changed","id":"49","r":"sensors","t":"event","uniqueid":"00:1f:ee:00:00:00:00:00-03-0006"} (ret = 16)
21:13:09:947 SC tick --> StateCallFunction
21:13:09:950 00:1f:ee:00:00:00:00:00-03-0006/config/mode expression: if (Item.val == 'momentary') { '01004841020006000D0306000206010D04060002' } else if (Item.val == 'rocker') { '01004841040006000D0306000206010D040600020600030306000206010304060002' } else if (Item.val.slice(0,7) == 'custom_') { '010048' + Item.val.slice(7, Item.val.length) } --> 01004841020006000D0306000206010D04060002
21:13:09:954 ZCL cmd attr 0x001FEE0000000000, ep: 0xE8, cl: 0xFC00, cmd: 0x02, mfcode: 0x0000, aps.id: 172, zcl.seq: 71
21:13:10:074 [INFO] - No button handler for: S2 (5502), unicast to: 0x0000, endpoint: 0xE8, cluster: 0xFC00, command: 0x04, payload: 00, zclSeq: 71

21:13:32:381 ZCL read attr 0x001FEE0000000000, ep: 0xE8, cl: 0xFC00, attr: 0x0001, mfcode: 0x0000, aps.id: 134, zcl.seq: 97
21:13:32:552 DEV Poll Busy /devices/0x001FEE0000000000 APS-DATA.confirm id: 134, ZCL seq: 97, status: 0x00
21:13:32:582 ZCL got data for node=0x2BD6, cl=0xFC00, at=0x0001, status=0x00, type=0x48
21:13:32:589 expression: var m = ''; for (var i = 4; i < ZclFrame.payloadSize; i++) { m += ('00' + ZclFrame.at(i).toString(16)).slice(-2); } m = m.toUpperCase(); if ( m == '41020006000D0306000206010D04060002') { Item.val = 'momentary'; } else if ( m == '41040006000D0306000206010D040600020600030306000206010304060002') { Item.val = 'rocker'; } else { Item.val = 'custom_' + m; } --> momentary
21:13:32:592 [INFO] - No button handler for: S2 (5502), unicast to: 0x0000, endpoint: 0xE8, cluster: 0xFC00, command: 0x01, payload: 0100004841020006000D0306000206010D04060002, zclSeq: 96
21:13:32:594 SC config/mode: synced
21:13:32:595 SC --> StateFinished
21:13:32:596 SC state change finished: 00:1f:ee:00:00:00:00:00-03-0006
```

[image]

[image]

steinbachd, 2024-01-02:
I can confirm it works - looks like the problem was in front of the computer, setting everything up from scratch fixed the issue. Thanks a lot for the quick turnaround, @SwoopX!