dresden-elektronik/deconz-rest-plugin #7463: Legrand Netatmo - 4 Scenes Wireless Remote Command with battery (077709L, 600091, 600191, 067755, 067756, 067757) #5021

Didier3L, 2023-12-18:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Good morning,

I updated Deconz to version 2.24.2.

My Legrand 067755 equipment which sent buttonevent 1002, 2002, 3002 and 4002 now sends 251998, 252998, 253998, 254998

### Steps to reproduce the behavior

```
{
    "102": {
        "config": {
            "battery": 0,
            "on": true,
            "reachable": true
        },
        "etag": "44f5ccb2d3dc3cfaae438e36326c1b95",
        "lastannounced": "2023-12-18T12:45:24Z",
        "lastseen": "2023-12-18T12:46Z",
        "manufacturername": "Legrand",
        "mode": 1,
        "modelid": "Wireless Scenes Command",
        "name": "Commande 4 sc\u00e8nes",
        "state": {
            "buttonevent": -254998,
            "lastupdated": "2023-12-18T12:45:39.545"
        },
        "swversion": "0024",
        "type": "ZHASwitch",
        "uniqueid": "00:04:74:00:00:92:b2:2e-01-0005"
    }
}
```

### Expected behavior

Comportement attendu

### Screenshots

Captures d'écran 
[image]

### Environment

Environnement 
[image]

### deCONZ Logs

Journaux deCONZ 

### Additional context

Contexte supplémentaire

Mimiix, 2023-12-18:
From what version did you update?

@smanar can you check out :)?

Didier3L, 2023-12-18:
> A partir de quelle version as-tu mis à jour ?
> 
> @Smanarpouvez-vous vérifier :)?

2.20.01 / 19/09/2022

Smanar, 2023-12-18:
Sure.

The code used is

`"eval": "Item.val = (ZclFrame.at(0) + 18) * -1000 + 2"`

Value was 

> 1 = -19 = 1002
> 2 = -20 = 2002
> 3 = -21 = 3002
> 4 = -22 = 4002

So idk how to obain -251998, -252998, -253998, -254998 ???
By deduction value are now 234 235 236 237

IDK what happen but the code work now with unsigned value. The hexa value was 0xEC, that give  236 in unsigned and -19 in signed.

Can try with this code pls 

`"eval": "Item.val = ((ZclFrame.at(0) << 24 >> 24) + 18) * -1000 + 2"`

Didier3L, 2023-12-18:
thank you Smanar
It works correctly

```
{
    "102": {
        "config": {
            "battery": 80,
            "on": true,
            "reachable": true
        },
        "etag": "5a1f2b3257b9052498db210fc74bf54b",
        "lastannounced": "2023-12-18T16:45:47Z",
        "lastseen": "2023-12-18T16:45Z",
        "manufacturername": "Legrand",
        "mode": 1,
        "modelid": "Wireless Scenes Command",
        "name": "Commande 4 sc\u00e8nes",
        "state": {
            "buttonevent": 3002,
            "lastupdated": "2023-12-18T16:46:09.187"
        },
        "swversion": "0024",
        "type": "ZHASwitch",
        "uniqueid": "00:04:74:00:00:92:b2:2e-01-0005"
    }
}
```

Smanar, 2023-12-18:
Nice.
PR done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7466

closed 2023-12-18 by manup