dresden-elektronik/deconz-rest-plugin #7470: Philips Hue Centris 2-spot ceiling lamp

Finne75, 2023-12-19:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Philips Hue Centris 2-spot ceiling lamp

### Manufacturer

Philips

### Model identifier

5061030P7_01, 5061030P7_02, 5061030P7_03

### Device type to add

Light

### Node info

The lamp is a compound of three individual lights, two spots and one downlight. The model ID of each of the lights is as variant of 5061030P7 with a suffix _01, _02_, _03. E.g., in my case, 5061030P7_01 is the downlight. Apart of the model ID, the clusters of all lights are absolutely identically as far as I can see. I did some experimentation and with [this](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/philips/light_zb3_C.json) DDF, but the effects etc. don't seem to work.

See also: https://www.zigbee2mqtt.io/devices/5061030P7.html

[image]

[image]

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]
[image]

### Further relevant clusters

### On/Off
[image]
[image]

### Level Control
[image]
[image]

### Color Control
[image]
[image]
[image]
[image]
[image]
[image]

### Hue Entertainment Cluster
[image]

ebaauw, 2023-12-19:
The _Hue Effects_ cluster is missing.  You need to update the lights's firmwares (all three nodes separately).

Finne75, 2023-12-19:
> The _Hue Effects_ cluster is missing. You need to update the lights's firmwares (all three nodes separately).

So I added one of them (_03) to the light_zb3_C.json DDF, and after reload, the Hue Effects Cluster appeared (see above). Can I update the firmware via deCONZ or do I need to use the Hue bridge?

ebaauw, 2023-12-19:
You should be able to upgrade them through the deCONZ OTA plugin, or over Bluetooth, using the Hue app without a Hue bridge.  They probably won't upgrade to the latest version immediately; the upgrade might take several turns, and the _Image Type_ might change (to 0x011F) during this process.  In the end, _ZCL Version_ should be 8, and I expect the _SW Build ID_ to become 1.104.2.

I would expect them to be compatible with `light_zb3_C.json` after the upgrade.

closed 2024-01-17 by github-actions[bot]