dresden-elektronik/deconz-rest-plugin #7473: Power meter support for WOOLLEY/eWeLink SmartPlug CK-BL702-SWP-01(7020)

larskoenig, 2023-12-20:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

WOOLLEY Zigbee Smart Plug

### Manufacturer

eWeLink

### Model identifier

CK-BL702-SWP-01(7020)

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

[image]

### Any other cluster of relevance/interest

larskoenig, 2023-12-20:
On/Off works fine but power metering does not show up. Here's a thread successfully adding support to zigbee2mqtt: https://github.com/Koenkk/zigbee2mqtt/issues/19334

Smanar, 2023-12-22:
Can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "eWeLink",
  "modelid": "CK-BL702-SWP-01(7020)",
  "vendor": "Immax Neo",
  "product": "WOOLLEY Zigbee Smart Plug",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/on",
          "refresh.interval": 360
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xFC11"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "read": {
            "at": "0x7006",
            "cl": "0xFC11",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x7006",
            "cl": "0xFC11",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          }
        },
        {
          "name": "state/current",
          "read": {
            "at": "0x7004",
            "cl": "0xFC11",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x7004",
            "cl": "0xFC11",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          }
        },
        {
          "name": "state/voltage",
          "read": {
            "at": "0x7005",
            "cl": "0xFC11",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x7005",
            "cl": "0xFC11",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}
```

You probably need some value convertion.

larskoenig, 2023-12-22:
Hi! Thanks so much for the DDF. Just tried it and it works! As you said, I had to scale voltage and power by 0.001, but other than that it looks fine.

Smanar, 2023-12-23:
Was lucky, If you can make a PR to submit the device officialy ?
Or share the final DDF for I make it for you.

larskoenig, 2023-12-23:
I've attached the json file with the conversions:
[ck-bl702-swp-01(7020).json](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13760231/ck-bl702-swp-01.7020.json)
What's still missing is power consumption though. I guess it's somehow related to cluster FC57, but I'm not sure how to analyze this further.

Smanar, 2023-12-24:
From Z2M this device don't support consumption, you have it in description ? Perhaps its just still missing ?

PR done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7488

larskoenig, 2023-12-25:
From the vendor's description of the device, I suppose it supports consumption, and in Z2M it might also still be missing. However, I don't really know how I can help to add support.

Smanar, 2023-12-26:
Right, I have read the complete post on Z2M and it seem it's present.
As report are native, you can try to get them in logs.
Can enable deconz log with flag "Info" and "Info_l2", thoses one are easy to understand.
If you see nothing can eanble "ZCL", in this one you will see the request for sure, but not easy to understand.

For me the request need to be a classic reporting on one of the 2 unknow cluster.

closed 2023-12-27 by manup