dresden-elektronik/deconz-rest-plugin #7476: bseed 3 gang switch tuya _TZ3000_xxxxxxxx TS0003

Onshuis, 2023-12-22:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

BSEED zigbee switch 3 gang Touch Smart

### Manufacturer

_TZ3000_eatarlvc

### Model identifier

TS0003

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

The "On/Off light 35" in the screen shit is the same device. It is an inwall 3-gang switch. When i push the switches, it all works independently. 
If i switch within deCONZ, all 3 switches, switch at the same time.

Onshuis, 2023-12-22:
Im happy though to see when i press the switch(light) from phoscon, only one gang is now operated instead of all 3!

Ok this seems the same as: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6682

So i changed the manufacturer and took the DDF json file inside :~/.local/share/dresden-elektronik/deCONZ/devices$
Restarted deconz and now i can individually switch the 3 gangs. 

But what to do next? Do i leave the DDF file, or what must i do?

Mimiix, 2023-12-22:
> But what to do next? Do i leave the DDF file, or what must i do?

If you can: Make a PR out of it :)

Onshuis, 2023-12-22:
No i dont know how, if i click new pull request it tells me to compare and everything is grayed out.

Mimiix, 2023-12-22:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-create-a-pull-request-to-submit-a-new-Device-Description-File

There :)

Onshuis, 2023-12-22:
Ok i did something, will do the same for the dimmer, thank you !

Onshuis, 2024-01-03:
Ill keep using the DDF in my .local directory, it works fine this way.

closed 2024-01-03 by Onshuis