dresden-elektronik/deconz-rest-plugin #7478: Lightsensor Luxmeter Zigbee Tuya

ProductEigel, 2023-12-22:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Lightsensor Luxmeter Zigbee Tuya

### Manufacturer

_TZE200_knx7nnka

### Model identifier

TS0601

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

ProductEigel, 2024-01-15:
Noch nicht gelöst

ProductEigel, 2024-02-07:
Noch nicht gelöst

Mimiix, 2024-02-07:
@Smanar can you help out :)?

Smanar, 2024-02-07:
Hello

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_khx7nnka", "_TZE200_yi4jtqq1"],
  "modelid": ["TS0601", "TS0601"],
  "product": "Tuya Lightsensor Luxmeter",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xEF00"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/dark"
        },
        {
          "name": "config/tholddark"
        },
        {
          "name": "config/tholdoffset"
        },
        {
          "name": "state/daylight"
        },
        {
          "name": "state/lightlevel"
        },
        {
          "name": "state/lux",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 2,
            "script": "../generic/illuminance_cluster/lux_to_lightlevel.js",
            "fn": "tuya"
          },
          "default": 0
        }
      ]
    }
  ]
}
```

Just some remark:
This device is powered ? because it seem it can work as router ? Have set the "sleeper" mode to false.
To work this DDF NEED to be in the devices/tuya folder, because of relative path  `"script": "../generic/illuminance_cluster/lux_to_lightlevel.js",` if you are on docker, you can edit the path and copy too the file `lux_to_lightlevel.js` in the same folder

ProductEigel, 2024-02-15:
@Smanar  
Hi I'm a bit confused right now :D what should I add where? I use Deconz as a container

ProductEigel, 2024-02-15:
edit: 
Image: deconzcommunity/deconz:latest

ProductEigel, 2024-02-15:
[image]

"script": "./lux_to_lightlevel.js",

doens't work...

[image]

ProductEigel, 2024-02-15:
i've tested also with a complett path, also with "/opt/deCONZ/devices/tuya/lux_to_lightlevel.js" or "/opt/deCONZ/devices/lux_to_lightlevel.js" or "/opt/deCONZ/lux_to_lightlevel.js"

ProductEigel, 2024-02-15:
okay, I have found the error. The file must be called json not js

Smanar, 2024-02-15:
You are using the official docker image ?

So when you make "ls -la" we need to see the *.js file and the *.json ?
And if both file are here you can just use

`"script": "lux_to_lightlevel.js",`

ProductEigel, 2024-02-15:
[image]
but now the "lux" wont changed ....

ProductEigel, 2024-02-15:
my mistake, I had named the DDF file "lux_to_lightlevel.js" :D and not after the device. I have now adjusted this so that there is no confusion. And yes I use the official Docker image, I just haven't found the "lux_to_lightlevel.js" yet and don't know the path either. I am an absolute newcomer to Deconz.

ProductEigel, 2024-02-15:
@Smanar so I stole the "lux_to_lightlevel.js" from a Windows installation, but the device is still listed as a temperature controller and the display does not work. It always remains at 0 lux.

Smanar, 2024-02-16:
>the device is still listed as a temperature controller

This is strange, can you take a look direclty in the API ? usig phoscon/help/API Information/sensors.

>It always remains at 0 lux

To make test withoutscript can you use instead
```
        {
          "name": "state/lux",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 2,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        }
```

You will need to  restart deconz after the change.

ProductEigel, 2024-02-16:
{
    "config": {
        "on": true,
        "reachable": true,
        "tholddark": 12000,
        "tholdoffset": 7000
    },
    "etag": "06e57be68d9573ef96c4ed74dc4527e4",
    "lastannounced": null,
    "lastseen": "2024-02-16T19:11Z",
    "manufacturername": "_TZE200_khx7nnka",
    "modelid": "TS0601",
    "name": "Lichtsensor",
    "state": {
        "dark": false,
        "daylight": false,
        "lastupdated": "2024-02-16T19:11:45.128",
        "lightlevel": 0,
        "lux": 3
    },
    "swversion": "1.0.6",
    "type": "ZHALightLevel",
    "uniqueid": "a4:c1:38:66:cf:78:e4:b0-01-ef00"
}

ProductEigel, 2024-02-16:
and Events:

{
    "20:11:45:161": {
        "e": "changed",
        "id": "35",
        "r": "sensors",
        "state": {
            "dark": false,
            "daylight": false,
            "lastupdated": "2024-02-16T19:11:45.128",
            "lightlevel": 0,
            "lux": 3
        },
        "t": "event",
        "uniqueid": "a4:c1:38:66:cf:78:e4:b0-01-ef00"
    },
    "20:11:59:706": {
        "e": "changed",
        "id": "35",
        "r": "sensors",
        "state": {
            "dark": false,
            "daylight": false,
            "lastupdated": "2024-02-16T19:11:59.656",
            "lightlevel": 0,
            "lux": 3
        },
        "t": "event",
        "uniqueid": "a4:c1:38:66:cf:78:e4:b0-01-ef00"
    },
    "20:12:14:253": {
        "attr": {
            "id": "35",
            "lastannounced": null,
            "lastseen": "2024-02-16T19:12Z",
            "manufacturername": "_TZE200_khx7nnka",
            "mode": 1,
            "modelid": "TS0601",
            "name": "Lichtsensor",
            "swversion": "1.0.6",
            "type": "ZHALightLevel",
            "uniqueid": "a4:c1:38:66:cf:78:e4:b0-01-ef00"
        },
        "e": "changed",
        "id": "35",
        "r": "sensors",
        "t": "event",
        "uniqueid": "a4:c1:38:66:cf:78:e4:b0-01-ef00"
    },
    "20:12:14:256": {
        "e": "changed",
        "id": "35",
        "r": "sensors",
        "state": {
            "dark": false,
            "daylight": false,
            "lastupdated": "2024-02-16T19:12:14.185",
            "lightlevel": 0,
            "lux": 3
        },
        "t": "event",
        "uniqueid": "a4:c1:38:66:cf:78:e4:b0-01-ef00"
    },
    "20:12:28:787": {
        "e": "changed",
        "id": "35",
        "r": "sensors",
        "state": {
            "dark": false,
            "daylight": false,
            "lastupdated": "2024-02-16T19:12:28.713",
            "lightlevel": 0,
            "lux": 3
        },
        "t": "event",
        "uniqueid": "a4:c1:38:66:cf:78:e4:b0-01-ef00"
    },
    "20:12:43:231": {
        "e": "changed",
        "id": "35",
        "r": "sensors",
        "state": {
            "dark": false,
            "daylight": false,
            "lastupdated": "2024-02-16T19:12:43.241",
            "lightlevel": 0,
            "lux": 3
        },
        "t": "event",
        "uniqueid": "a4:c1:38:66:cf:78:e4:b0-01-ef00"
    }
}

ProductEigel, 2024-02-16:
[image]

ProductEigel, 2024-02-16:
@Smanar the light is on, so it cannot be 0 lux

ProductEigel, 2024-02-16:
@Smanar okay all commands back, it reacts now, although in my opinion with too little, it is displayed at full light 37Lux

ProductEigel, 2024-02-16:
@Smanar thank you very much for your support :) the ticket can be closed :) have a nice weekend everyone ;)

ProductEigel, 2024-02-16:
completed

Smanar, 2024-02-17:
Lol, you was too fast for me.
On my last try the DDF just take a raw value and display it at lux and no more, so nothing in state/lightlevel for exemple.

Idaho947, 2024-04-24:
Hello, can we reopen the ticket ? I've too off this sensor and still see as a plug...

Smanar, 2024-04-24:
It's same manufacture name ? _TZE200_khx7nnka
Else better to open a new issue, or just add the new manufactureName at previous DDF to test if there are clone.

Idaho947, 2024-04-24:
yes, it's exactly the same.
As the ddf isn't in a stable version and in view of your last message, I thought the integration wasn't complete.

Smanar, 2024-04-24:
Oups sorry, my bad, have forget this device

Smanar, 2024-04-24:
So have checked again, the first one need to work, it's a raw value for lux that can be converted for other field
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_khx7nnka", "_TZE200_yi4jtqq1"],
  "modelid": ["TS0601", "TS0601"],
  "product": "Tuya Lightsensor Luxmeter",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xEF00"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/dark"
        },
        {
          "name": "config/tholddark"
        },
        {
          "name": "config/tholdoffset"
        },
        {
          "name": "state/daylight"
        },
        {
          "name": "state/lightlevel"
        },
        {
          "name": "state/lux",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 2,
            "script": "../generic/illuminance_cluster/lux_to_lightlevel.js",
            "fn": "tuya"
          },
          "default": 0
        }
      ]
    }
  ]
}
```

With same requirement

> To work this DDF NEED to be in the devices/tuya folder, because of relative path "script": "../generic/illuminance_cluster/lux_to_lightlevel.js", if you are on docker, you can edit the path and copy too the file lux_to_lightlevel.js in the same folder

On your last try was just a "hack" to test code", so not good values.

Idaho947, 2024-04-24:
Yes everything is fine ! It works perfectly.

Smanar, 2024-04-24:
Wana make the DDF or want I make it ?

>turn of the bluelight

On the device ? I don't see something about that on this device ...

Idaho947, 2024-04-25:
Yes... I put a scotch tape on it 😋

Smanar, 2024-04-25:
Lool, I was searching zigbee features, there is guys doing same at my work for alarm "indicator light".

BTW do you want to submit the DDF, to have it officialy ?

Idaho947, 2024-04-25:
Yes of course

Smanar, 2024-04-26:
But do you want to make it yourself or want I make it ?

Idaho947, 2024-04-26:
You can.
I will resend them to the seller in fact. The are ok when the lux is not 0. If the senser is in the dark, it send several message per second...

Smanar, 2024-04-26:
if state/lux is 0 ?
Do you still have it ? It's possible to add a "filter" on the DDF. For exemple send update only if the value is different or > at a value.

Idaho947, 2024-04-26:
Ok I can try

Smanar, 2024-04-27:
You have a "spam" capture ?
To see what is the spammed request, and how to clock it ?

For exemple if it happen only if the state/lux  = 0 can use a check with the 0 value if it's just a spamming with the same value, can add a check that compare with prevous value and ignore it if it haven"t changed.

This code for exemple will work I think, replace

```
        {
          "name": "state/lux",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 2,
            "script": "../generic/illuminance_cluster/lux_to_lightlevel.js",
            "fn": "tuya"
          },
          "default": 0
        }
```

by
```

        {
          "name": "state/lux",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 2,
            "script": "{"fn": "tuya", "dpid": 2, "eval": "if R.item('state/valve').val != Attr.val {Item.val = Attr.val;}"}",
            "fn": "none"
          },
          "default": 0
        }
```

It's just a sample, it will update only state/lux with raw value, so "state/lightlevel" for exemple will stop working.
but if it work, it will block the  spam

Idaho947, 2024-04-27:
It spam a lot when lux = 0 as you see in this logs :

`{
    "12:32:20:784": {
        "e": "changed",
        "id": "657",
        "r": "sensors",
        "state": {
            "dark": true,
            "daylight": false,
            "lastupdated": "2024-04-27T10:32:21.905",
            "lightlevel": 0,
            "lux": 0
        },
        "t": "event",
        "uniqueid": "a4:c1:38:9d:17:26:47:5c-01-ef00"
    },
    "12:32:21:426": {
        "e": "changed",
        "id": "657",
        "r": "sensors",
        "state": {
            "dark": true,
            "daylight": false,
            "lastupdated": "2024-04-27T10:32:22.512",
            "lightlevel": 0,
            "lux": 0
        },
        "t": "event",
        "uniqueid": "a4:c1:38:9d:17:26:47:5c-01-ef00"
    },
    "12:32:21:991": {
        "e": "changed",
        "id": "657",
        "r": "sensors",
        "state": {
            "dark": true,
            "daylight": false,
            "lastupdated": "2024-04-27T10:32:23.118",
            "lightlevel": 0,
            "lux": 0
        },
        "t": "event",
        "uniqueid": "a4:c1:38:9d:17:26:47:5c-01-ef00"
    },
    "12:32:22:849": {
        "e": "changed",
        "id": "657",
        "r": "sensors",
        "state": {
            "dark": true,
            "daylight": false,
            "lastupdated": "2024-04-27T10:32:23.723",
            "lightlevel": 0,
            "lux": 0
        },
        "t": "event",
        "uniqueid": "a4:c1:38:9d:17:26:47:5c-01-ef00"
    },
    "12:32:23:202": {
        "e": "changed",
        "id": "657",
        "r": "sensors",
        "state": {
            "dark": false,
            "daylight": false,
            "lastupdated": "2024-04-27T10:32:24.329",
            "lightlevel": 14625,
            "lux": 29
        },
[... 357 lines cut ...]
        "state": {
            "dark": false,
            "daylight": false,
            "lastupdated": "2024-04-27T10:32:52.118",
            "lightlevel": 13980,
            "lux": 25
        },
        "t": "event",
        "uniqueid": "a4:c1:38:9d:17:26:47:5c-01-ef00"
    },
    "12:32:51:685": {
        "e": "changed",
        "id": "657",
        "r": "sensors",
        "state": {
            "dark": false,
            "daylight": false,
            "lastupdated": "2024-04-27T10:32:52.724",
            "lightlevel": 14625,
            "lux": 29
        },
        "t": "event",
        "uniqueid": "a4:c1:38:9d:17:26:47:5c-01-ef00"
    }
}`

Smanar, 2024-04-27:
It spam less with other value, but still spamming

```
  "12:32:23:202": {
    "e": "changed",
    "id": "657",
    "r": "sensors",
    "state": {
      "dark": false,
      "daylight": false,
      "lastupdated": "2024-04-27T10:32:24.329",
      "lightlevel": 14625,
      "lux": 29
    },
    "t": "event",
    "uniqueid": "a4:c1:38:9d:17:26:47:5c-01-ef00"
  },
  "12:32:26:40": {
    "e": "changed",
    "id": "657",
    "r": "sensors",
    "state": {
      "dark": false,
      "daylight": false,
      "lastupdated": "2024-04-27T10:32:26.863",
      "lightlevel": 14625,
      "lux": 29
    },
    "t": "event",
    "uniqueid": "a4:c1:38:9d:17:26:47:5c-01-ef00"
  },
```

>>With your new ddf, the lux doesn't change

Arf fuck, my fault, it's a copy/paste, can you replace

`"eval": "if R.item('state/valve').val != Attr.val {Item.val = Attr.val;}",`

by

`"eval": "if R.item('state/lux').val != Attr.val {Item.val = Attr.val;}",`

Idaho947, 2024-04-27:
Not better with this ddf :

`{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_khx7nnka", "_TZE200_yi4jtqq1"],
  "modelid": ["TS0601", "TS0601"],
  "product": "Tuya Lightsensor Luxmeter",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xEF00"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/dark"
        },
        {
          "name": "config/tholddark"
        },
        {
          "name": "config/tholdoffset"
        },
        {
          "name": "state/daylight"
        },
        {
          "name": "state/lightlevel"
        },
        {
          "name": "state/lux",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 2,
            "eval": "if R.item('state/lux').val != Attr.val {Item.val = Attr.val;}",
            "fn": "tuya"
          },
          "default": 0
        }
      ]
    }
  ]
}`

Smanar, 2024-04-27:
I don't remember if you have access to the complete application ?
Perhaps something visible in log deconz/help/debug View with flag "DDF". Because all seem fine for me.

Smanar, 2024-04-27:
I have made a synthax error, but I don't see my error, can you try with
`"eval": "if (R.item('state/lux').val != Attr.val) {Item.val = Attr.val;}",`

Idaho947, 2024-04-27:
With you last code, the sensor spam when the lux = 0

Idaho947, 2024-04-27:
It works but the change of the lux entitie is very long more tha 1 min.

The tuya sensor is not very compliant with the zigbee network

Smanar, 2024-04-27:
But you have 2 devices like this one ?
the a4:c1:38...... is using the new DDF but the 04:cf:8c........ is still using another one.

>In HA the value doesn't change :

It's possible they are using state/lightlevel instead of state/lux ? Because they are using same API, not possible having different value on phoscon and HA.

On your log you have "-->undefined" probably because the value is ignored (not changed), it work because we can see lux=26 on the API, and just before it have worked for the 12 value.

But yes you still have lot of traffic, you are sure it's not the other device ? The 04:cf:8c........ ?
Can you check if you still have the "spam" on log ?

If the DDF is working (no more spam), yes, you can return thoses devices, not possible to set the reporting on this device, have checked Z2M too, if ignoring unchanged value don't work, I don't see what we can do ....

Idaho947, 2024-04-27:
I have too of those device but delete the second one via the guid and reboot dexconz. I dont understand why it is already here.
Before the last change everything work in ha despite the spam.
The tuya is as the other of rhis brand cheap and create a lot of issue...

Smanar, 2024-04-27:
But both are using the same DDF ? It's not possible, you haven't 2 similars device but using 2 different DDF ?

Idaho947, 2024-04-28:
No I ve only one in my network. I dont understand what is the second one...

Smanar, 2024-04-28:
It's a Xiaomi one according the mac adress.
But just remove the tuya one, look at traffic, and re-include it, to be sure it's the one that provole that.
And you still have the "spam" on logs ?

Idaho947, 2024-04-28:
Yes I remove it and everything is fine. I reinclude it and the spam is here.

Smanar, 2024-04-28:
And you still have spam like here

```
  "12:32:21:426": {
    "e": "changed",
    "id": "657",
    "r": "sensors",
    "state": {
      "dark": true,
      "daylight": false,
      "lastupdated": "2024-04-27T10:32:22.512",
      "lightlevel": 0,
      "lux": 0
    },
    "t": "event",
    "uniqueid": "a4:c1:38:9d:17:26:47:5c-01-ef00"
  },
  "12:32:21:991": {
    "e": "changed",
    "id": "657",
    "r": "sensors",
    "state": {
      "dark": true,
      "daylight": false,
      "lastupdated": "2024-04-27T10:32:23.118",
      "lightlevel": 0,
      "lux": 0
    },
```

Same value, in less than 1 s delay ?

This should not appear with the new DDF ...

closed 2024-05-27 by github-actions[bot]