dresden-elektronik/deconz-rest-plugin #7479: innr SP 240 smart plug

baltena, 2023-12-22:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

SP 240 smart plug

### Manufacturer

innr

### Model identifier

SP240

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering
[image]

### Electrical Measurement
[image]
[image]
[image]
[image]
[image]

### Any other cluster of relevance/interest
Tuya sepecific

Smanar, 2023-12-22:
I m a bit affraid by the presence of the tuya cluster but have you tried a classic DDF like this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/innr/sp_120.json ?

With editing the modelid ?

sven2c, 2023-12-23:
Hello,
I'm able to use the plug but just ON/OFF functionallity.
I'm using the API to control the plug via Node Red, but I'm not able to read out the Power Data.
Anyhow I'm including this in the DDF. 
I can see the Data working in the Deconz GUI, but the Phoscon RestAPI will not show up the sensor data

Smanar, 2023-12-23:
There is a PR in waiting list for this device https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7486

closed 2023-12-27 by manup