dresden-elektronik/deconz-rest-plugin #7484: Deconz (Conbee III) no longer recognize Aqara Smart Radiator Thermostat E1

Hanssie66, 2023-12-23:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I saw that deCONZ no longer recognize my Aqara Smart Radiator Thermostat E1 as a thermostat. At first I thought I had a problem with pairing. However, an in-depth analysis taught me that the thermostat did pair with the CONZ, but was not recognized as a thermostat. At first the thermostat was recognized as described [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6351). However, after a reset, the thermostat was no longer recognized as a thermostat (a lot of information had disappeared). How can I solve this issue? Or is this problem part of the deCONZ software?

[image]

### Steps to reproduce the behavior

1: Put thermosthat in pairing mode
2: select Thermostats and activate the button to Connect new thermostat
3: notting connected, however the pairing tree shows that the thermostat is paired 

### Expected behavior

The thermostat should be paired and recognized as thermostat and not as an unknown device. 

### Environment

- Host system: (Raspberry Pi 3B+)
- Running method: deCONZ Add-on
- Firmware version: (264D0900)
- deCONZ version (not Home assistant Addon version!): (2.24.2)
- Device:  ConBee III /
- Do you use an USB extension cable: yes 
- Is there any other USB or serial devices connected to the host system? No

SwoopX, 2023-12-24:
Can you please retry with deconz v2.25.0?

closed 2024-01-22 by github-actions[bot]

mendes588, 2024-02-24:
> ### Does the issue really belong here?
>
>     * [x]  I definitively want to report a bug within deCONZ or its REST-API
>
>
> ### Is there already an existing issue for this?
>
>     * [x]  I have searched [...]

I can confirm the issue. It show's up on zigbee2mqtt but no values are shared and on the TRV itself the "connected" icon doesn't stay on (connected)