dresden-elektronik/deconz-rest-plugin #7489: Connection issue with Conbee III and Aqara contact sensor E1 (MCCGQ14LM)

Gloomyeye, 2023-12-24:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I switch recently from conbee II to conbee III stick. Now, all my Aqara contact sensor E1 (MCCGQ14LM) have connection issues. I tried to repair the devices without deleting the device in phoscon. This way was not working. No sensor update was retrieved successfully. Another idea was to delete the device and repair is from scratch. However, it is not possible anymore to connect my Aqara contact sensor E1. The device is not showing up. I changed the location (1m to the conbee III stick), changed the battery, tried to push every second the button while in pairing mode. Nothing helped.

I installed the lastest firmware on the conbee III stick. The environment with the conbee II stick was working without any issues or changes.

### Steps to reproduce the behavior

1: Put contact sensor E1 in pairing mode
2: Select sensors and activate the button to Connect new sensors
3: Nothing connected, no device shows up

### Expected behavior

Already paired sensors sends correct sensor values. A repair of the device is possible.

### Environment

- Host system: Raspberry Pi 4B
- Running method: Docker container
- Firmware version: 264e0900
- deCONZ version (not Home assistant Addon version!): 2.25.0-beta
- Device: ConBee III
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? no

Smanar, 2023-12-28:
Hello, perhaps this PR will help you ?
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7497

Gloomyeye, 2023-12-28:
I will try with the new release.

Smanar, 2023-12-29:
Next one, this PR will be merged in the v2.25.1

closed 2024-01-01 by Gloomyeye