dresden-elektronik/deconz-rest-plugin #7490: New version of Hue Wall Switch (RDM004, 9290030171A)

koenheemskerk, 2023-12-24:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Philips Hue Wall Switch (2023)

### Manufacturer

Philips

### Model identifier

RDM004

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

-

### Basic

-

Mimiix, 2023-12-24:
Can you please add all screenshots?

ebaauw, 2023-12-24:
Where did you get it?  I cannot seem to find any mention of a new model.  Particularly interested to know if this model does support hybrid mode (as the previous model doesn’t).

koenheemskerk, 2023-12-29:
Added the screens!

Syorn630, 2023-12-31:
I've bought two of these RDM004 Philips hue wall switch modules this week as well and faced the same issue when trying to connect them to deconz via the Phoscon web app. Searching for a solution finally brought me here.

Has anybody an idea how to get the new version working? At least a quick fix? I've tried to copy the DDF file of the RDM001 version and just changed the Model ID. This got me one step further: the switch module now shows up in Phoscon but unfortunately it doesn't transmit anything when the switch is toggled.

When looking at the API events, every 5 minutes the following two events are received:

```
"16:26:27:436": {
    "attr": {
        "id": "6",
        "lastannounced": "2023-12-31T15:26:27Z",
        "lastseen": "2023-12-31T15:25Z",
        "manufacturername": "Signify Netherlands B.V.",
        "modelid": "RDM004",
        "name": "Switch 6",
        "productname": "Hue wall switch module",
        "swversion": "2.60.16",
        "type": "ZHASwitch",
        "uniqueid": "00:17:88:01:xxxx"
    },
    "e": "changed",
    "id": "6",
    "r": "sensors",
    "t": "event",
    "uniqueid": "00:17:88:01:xxxx"
},
"16:26:27:437": {
    "config": {
        "battery": 100,
        "devicemode": "singlerocker",
        "on": true,
        "reachable": true
    },
    "e": "changed",
    "id": "6",
    "r": "sensors",
    "t": "event",
    "uniqueid": "00:17:88:01:xxxx"
}
```

Additionally, if the switch was toggled, the first of the two event is received, but not more than once per minute.

Any ideas would be greatly appreciated.

closed 2024-01-30 by github-actions[bot]

ebaauw, 2024-09-28:
> Particularly interested to know if this model does support hybrid mode (as the previous model doesn’t).

Yes, it does, see #7947.