dresden-elektronik/deconz-rest-plugin #7492: Profalux ZigBee roller shutters

BEskandari, 2023-12-25:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Profalux ZigBee roller shutters

### Manufacturer

Profalux

### Model identifier

unknown

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Identify
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### On/off
[image]
[image]

### Level Control
[image]
[image]

### Time
[image]

### Commissioning
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Shade Configuration
[image]

BEskandari, 2023-12-25:
The device was configured without DDF in previous version by @Smanar https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2739

Smanar, 2023-12-26:
It seem he have a strange version, without manufacture name and without Modelid.
If it's not a bug, not possible to include this model. Deconz need them to reconise the device.

BEskandari, 2023-12-26:
Thanks, I have found this on other communities:
"manufId":4368,"manufName":"Profalux"
But I have to check with the vendor to see if there is an issue on that.

Smanar, 2023-12-26:
Oups, @BEskandari is right I have just see a "hack" in the code to support this device, with missing information.

```
        //Add missing values for Profalux device
        if (existDevicesWithVendorCodeForMacPrefix(node->address(), VENDOR_PROFALUX))
        {
            //Shutter ?
            if (i->deviceId() == DEV_ID_ZLL_COLOR_LIGHT)
            {
                lightNode.setManufacturerName(QLatin1String("Profalux"));
                lightNode.setModelId(QLatin1String("PFLX Shutter"));
                lightNode.setNeedSaveDatabase(true);
            }
        }
```

So to work the device need to have
- Manufacture number 0x1110
- Device type DEV_ID_ZLL_COLOR_LIGHT
- Router device.

But this code is perhaps disabled with DD core, you can try to disable it, time to include the device.

BEskandari, 2023-12-26:
Thanks @Smanar , but this is beyond my knowledge, I will follow your instructions in case to check the DDF and make it work.

BEskandari, 2023-12-27:
> Just disable the DDF core, include your device with the legacy core, then re-enable DDF.

[image]

I don't see the option to disable DDF as mentioned in previous post by you by selecting "Draft". And last but not least how can I be sure that the node is included with the legacy code. Because now I have it in the deCONZ-gui but no mention in the API's.

closed 2023-12-27 by SwoopX