dresden-elektronik/deconz-rest-plugin #7494: Blind roller _TZE200_libht6ua TS0601

TonKro, 2023-12-27:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tuya brand zigbee blind motor 

### Manufacturer

_TZE200_libht6ua

### Model identifier

TS0601

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

There is a js version available for this device for zigbee2mqtt

https://github.com/javilopezalarcon/Blind-motor-_TZE200_libht6ua-z2m/blob/main/README.md

Mimiix, 2023-12-27:
Please add all screenshots

TonKro, 2023-12-27:
Can you help me what is missing and needed?

Mimiix, 2023-12-27:
Please add the screenshots in the first post.

TonKro, 2023-12-28:
Tnx, all pictures are brought to the inititial request.

Mimiix, 2024-01-04:
> Is there any expectation to make? Or is Zigbee2MQTT or ZHA a better solution?

Someone needs to help.

Nevertheless, it has been holidays and a lot of people are off.

closed 2024-02-03 by github-actions[bot]