dresden-elektronik/deconz-rest-plugin #7500: Ikea Parasoll

hyper2910, 2023-12-29:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Parasoll

### Manufacturer

Ikea

### Model identifier

E2013

### Device type to add

None

### Node info

[image]

### Endpoints and clusters

[image]
[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

hyper2910, 2023-12-29:
[image]
[image]
[image]

Mimiix, 2023-12-29:
Hi,

Clearly you didn't search as it's a duplicate of #7417 

Closing

closed 2023-12-29 by Mimiix