dresden-elektronik/deconz-rest-plugin #7507: parasoll mqtt no reading available open/close

hyper2910, 2024-01-02:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

no status change via mqtt

STATE is evrytime INITIALIZED

as Reading also only the following coming

Readings
IODev | deCONZ | 2023-12-31 12:46:27 -- | -- | -- attrTemplateVersion | Xiaomi_Aqara_MCCGQ11LM_Window_Door_Sensor_20211015 | 2023-12-31 13:06:25 battery | 94 | 2023-12-31 13:03:26 batteryPercent | 94 | 2023-12-31 13:03:26 reachable | 1 | 2023-12-31 13:03:26

### Steps to reproduce the behavior

.

### Expected behavior

a status/reading with open/close

### Screenshots

.

### Environment

- Host system: (Raspberry Pi / PC / NAS / MAC)
- Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
- Firmware version: (26xxyy00)
- deCONZ version (not Home assistant Addon version!): (2.xx.yy)
- Device: (ConBee I / ConBee II / ConBee III / RaspBee I / RaspBee II)
- Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?

### deCONZ Logs

.

### Additional context

.

hyper2910, 2024-01-02:
also no "wert" in P
[image]
hoscon

Mimiix, 2024-01-02:
Hi,

what does the RestAPI / Websocket show?

hyper2910, 2024-01-02:
Hi, 
what do you mean?

in phoscon also no "Status/WERT" is available.

[image]

Mimiix, 2024-01-02:
Hi,

You can check in Phoscon under Help > API information. There, filter for Sensors and select your Parasoll sensor.

If you open/close it, does it show changes under Events tab? Can you also show the Rest API output?

hyper2910, 2024-01-02:
Sensor Open:
```
{
    "config": {
        "battery": 94,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 2,
    "etag": "cfc8a9095086bb68996bf611ce4ebb33",
    "lastannounced": null,
    "lastseen": "2024-01-02T08:46Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "PARASOLL Door/Window Sensor",
    "name": "OpenClose_test",
    "state": {
        "lastupdated": "none",
        "lowbattery": null,
        "open": null
    },
    "swversion": "1.0.19",
    "type": "ZHAOpenClose",
    "uniqueid": "04:87:27:ff:fe:b8:a7:1c-02-0500"
}
```
Sensor Closed:
`{
    "config": {
        "battery": 94,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 2,
    "etag": "cfc8a9095086bb68996bf611ce4ebb33",
    "lastannounced": null,
    "lastseen": "2024-01-02T08:46Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "PARASOLL Door/Window Sensor",
    "name": "OpenClose_test",
    "state": {
        "lastupdated": "none",
        "lowbattery": null,
        "open": null
    },
    "swversion": "1.0.19",
    "type": "ZHAOpenClose",
    "uniqueid": "04:87:27:ff:fe:b8:a7:1c-02-0500"
}`

Unter Events:
{}
[image]

Mimiix, 2024-01-02:
Can you please fill the details of the template correctly? 

Also, can you show me a screenshot of the basic cluster of the device in deconz?

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

hyper2910, 2024-01-02:
[image]
[image]
[image]

Mimiix, 2024-01-02:
Still waiting on the template info.

hyper2910, 2024-01-02:
sorry my first time in github.

What do you need more?

Mimiix, 2024-01-02:
The template asks for environment details. You need to fill those.

hyper2910, 2024-01-02:
where did i get the template back?

Mimiix, 2024-01-02:
> where did i get the template back?

Edit your first post.

hyper2910, 2024-01-02:
> > where did i get the template back?
> 
> Edit your first post.

Done

Mimiix, 2024-01-02:
didyou try to re-pair the sensor already?

hyper2910, 2024-01-02:
> didyou try to re-pair the sensor already?

Yes,  done a few minutes ago and same issues

Mimiix, 2024-01-02:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

You might need to update the firmware on the conbee as it is outdated.

Mimiix, 2024-01-02:
Can you:

Remove the device,
Restart deconz
Start logging with log levels APS, APS L2, ERROR, ERROR L2, INFO, INFO L2. You can log in deconz under help , debug logs.
Start searching for new devices.
After pairing share the logs via Pastebin

hyper2910, 2024-01-02:
aftre repairing the "Status" is available.
many thanks.

closed 2024-01-02 by hyper2910

Mimiix, 2024-01-02:
Happy to hear it works. Take care :)

tkoenig, 2024-01-18:
I'm having the same issues with this device. 
I tried the same steps with removing the device and restarting and pairing it again but no luck.

@hyper2910 does it now show anything under "Werte" for you now?
@Mimiix I tried pairing this device before upgrading to the beta release, could this cause an issue?
@ebaauw I noticed from another issue that you also seem to have this device, does it work for you? (I'm using `homebridge-deconz`, where the device shows up, but also no state changes).

Below is what I get in the AI Information (with a little bit of a lag):

<details>
<summary>Events</summary>

```
{
    "20:25:16:161": {
        "attr": {
            "id": "28",
            "lastannounced": "2024-01-18T19:24:16Z",
            "lastseen": "2024-01-18T19:25Z",
            "manufacturername": "IKEA of Sweden",
            "mode": 1,
            "modelid": "PARASOLL Door/Window Sensor",
            "name": "PARASOLL Door/Window Sensor",
            "swversion": "1.0.19",
            "type": "ZHAOpenClose",
            "uniqueid": "3c:2e:f5:ff:fe:42:4c:f9-02-0500"
        },
        "e": "changed",
        "id": "28",
        "r": "sensors",
        "t": "event",
        "uniqueid": "3c:2e:f5:ff:fe:42:4c:f9-02-0500"
    },
    "20:26:16:476": {
        "attr": {
            "id": "28",
            "lastannounced": "2024-01-18T19:24:16Z",
            "lastseen": "2024-01-18T19:26Z",
            "manufacturername": "IKEA of Sweden",
            "mode": 1,
            "modelid": "PARASOLL Door/Window Sensor",
            "name": "PARASOLL Door/Window Sensor",
            "swversion": "1.0.19",
            "type": "ZHAOpenClose",
            "uniqueid": "3c:2e:f5:ff:fe:42:4c:f9-02-0500"
        },
        "e": "changed",
        "id": "28",
        "r": "sensors",
        "t": "event",
        "uniqueid": "3c:2e:f5:ff:fe:42:4c:f9-02-0500"
    },
    "20:27:33:893": {
        "attr": {
            "id": "28",
            "lastannounced": "2024-01-18T19:24:16Z",
            "lastseen": "2024-01-18T19:27Z",
            "manufacturername": "IKEA of Sweden",
            "mode": 1,
            "modelid": "PARASOLL Door/Window Sensor",
            "name": "PARASOLL Door/Window Sensor",
            "swversion": "1.0.19",
            "type": "ZHAOpenClose",
            "uniqueid": "3c:2e:f5:ff:fe:42:4c:f9-02-0500"
        },
        "e": "changed",
        "id": "28",
        "r": "sensors",
        "t": "event",
        "uniqueid": "3c:2e:f5:ff:fe:42:4c:f9-02-0500"
    },
    "20:28:34:208": {
        "attr": {
            "id": "28",
            "lastannounced": "2024-01-18T19:24:16Z",
            "lastseen": "2024-01-18T19:28Z",
            "manufacturername": "IKEA of Sweden",
            "mode": 1,
            "modelid": "PARASOLL Door/Window Sensor",
            "name": "PARASOLL Door/Window Sensor",
            "swversion": "1.0.19",
            "type": "ZHAOpenClose",
            "uniqueid": "3c:2e:f5:ff:fe:42:4c:f9-02-0500"
        },
        "e": "changed",
        "id": "28",
        "r": "sensors",
        "t": "event",
        "uniqueid": "3c:2e:f5:ff:fe:42:4c:f9-02-0500"
    },
    "20:34:18:966": {
        "attr": {
            "id": "28",
            "lastannounced": "2024-01-18T19:24:16Z",
            "lastseen": "2024-01-18T19:34Z",
            "manufacturername": "IKEA of Sweden",
            "mode": 1,
            "modelid": "PARASOLL Door/Window Sensor",
            "name": "PARASOLL Door/Window Sensor",
            "swversion": "1.0.19",
            "type": "ZHAOpenClose",
            "uniqueid": "3c:2e:f5:ff:fe:42:4c:f9-02-0500"
        },
        "e": "changed",
        "id": "28",
        "r": "sensors",
        "t": "event",
        "uniqueid": "3c:2e:f5:ff:fe:42:4c:f9-02-0500"
    }
}
```
</details>
<details>
<summary>RestAPI</summary>

```
{
    "config": {
        "battery": 0,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 2,
    "etag": "470548709c94e4c049515398cfd53fdd",
    "lastannounced": "2024-01-18T19:24:16Z",
    "lastseen": "2024-01-18T19:34Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "PARASOLL Door/Window Sensor",
    "name": "PARASOLL Door/Window Sensor",
    "state": {
        "lastupdated": "none",
        "lowbattery": null,
        "open": null
    },
    "swversion": "1.0.19",
    "type": "ZHAOpenClose",
    "uniqueid": "3c:2e:f5:ff:fe:42:4c:f9-02-0500"
}
```
</details