dresden-elektronik/deconz-rest-plugin #7508: Aqara T1 Pro (lumi.remote.cagl02) DDF not working

ChrisHae, 2024-01-02:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

After pairing the Aqara T1 pro cube it does not send any button events via Rest API. Also the endpoints and clusters in deCONZ gui are looking wrong (compared to older aqara cube). Its state can not be read via deCONZ.

### Steps to reproduce the behavior

Pairing the Aqara T1 pro cube with Phoscon App or in deCONZ direkctly.

### Expected behavior

Display of endpoints 2 and 3 and Cluster 0012 (Multistate Input) and 000C (Analog Input) in deCONZ gui.
Display of button events over web socket or Rest API

### Screenshots

Left: old cube right: Aqara T1 Pro
[image]
Rest API:
[image]

### Environment

- Host system: PC
- Running method: Windows
- Firmware version: (26780700)
- deCONZ version: 2.25.1
- Device: ConBee II
- Do you use an USB extension cable: Yes
- Is there any other USB or serial devices connected to the host system? No

Mimiix, 2024-01-02:
Hi Chris, 

Are you sure the device is paired correctly? Looks like your missing some clusters indicating not fully paired:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5732

ChrisHae, 2024-01-02:
You are right. Seems the failure was on my side. I had those pairing issues for some time and now double checked my setup. Seems that the USB extension caused the problems. So false alarm. problem solved.

closed 2024-01-02 by ChrisHae