dresden-elektronik/deconz-rest-plugin #7512: LIDL SilverCrest Smart plug

Brian-lub, 2024-01-03:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

LIDL SilverCrest Smart plug

### Manufacturer

_TZ3000_nkcobies

### Model identifier

TS011F

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Brian-lub, 2024-01-10:
Please let me know if you need more, i will be happy to help and test the device

Mimiix, 2024-01-10:
@Smanar can you check :)?

Smanar, 2024-01-11:
But I m seing this device in the DDf list https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/lidl/hg08673.json

Why deconz don't pick it ? what is your deconz version ?

Brian-lub, 2024-01-14:
Im using HA addon deConz Current version: 6.22.0

[image]

Smanar, 2024-01-14:
Can you take a look in phoscon/gateway to see wich one deconz version they are using ?
This device was enabled (gold status) in version v2.25.0-beta

Brian-lub, 2024-01-14:
[image]

Smanar, 2024-01-14:
You can see it in phscon too ^^, perhaps easier than lanching the GUI ?

But you haven't a recent enought version.
You can probably turn the status to "Gold" but at next restart it will return to "silver".
Perhpas in deconz setting on tab "control" lower the DDF level to "gold" from "Silver". On previous version, 2 month old , the DDF was "silver", but honnestly I havent checked the DDF state on 2.22 version (we are ATM on 2.25)

Brian-lub, 2024-01-14:
i have state draft in DDF

Mimiix, 2024-01-14:
> i have state draft in DDF. Do you have any influence on the HA add on?

Closing as it's a duplicate

closed 2024-01-14 by Mimiix