dresden-elektronik/deconz-rest-plugin #7518: modkam.ru    DIYRuZ_Flower    moisture sensor

ma-gu-16, 2024-01-06:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5209

Is it possible to add this sensor ? 

currently it doesnt work. 

Conbee2 Stick on iobroker 
Gateway Version: 2.24.2

### Steps to reproduce the behavior

[image]

[image]

### Expected behavior

all available data in zigbee

### Environment

server
debian deconz.

Gateway 2.24.2
Firmware 26580700
- Device: Conbee II

Smanar, 2024-01-06:
Hello, can you share a node capture from the GUI ?
To be sure all cluster are enabled, it's a custom sensor, you can select wich one cluster to use.

ma-gu-16, 2024-01-06:
thank you. i switched to zigbee app on iobroker and now i have all data. for me personally there is no further need.

closed 2024-01-06 by ma-gu-16