dresden-elektronik/deconz-rest-plugin #7525: Support for EasyAccessTouch door lock

cortado, 2024-01-09:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

EasyAccess EasyCodeTouch

### Manufacturer

EasyAccess / Nimly 

### Model identifier

EasyCodeTouch

### Device type to add

Lock

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

cluster 0xFEA2 seems to be the only difference from EasyFIngerTouch DDF?

Also seems to be some info that should be in an attribute somewhere?  at least seems to be missing for Phoscon app:
[image]

PS! Existing EasyFingerTOuch DDF states Onesti Products AS as manufacturer, that might be right for hardware or access module?  I only have info about EasyAccess/Nimly which is the Norwegian Company behind the product

Smanar, 2024-01-09:
From Z2M all thoses devices are same
`zigbeeModel: ['easyCodeTouch_v1', 'EasyCodeTouch', 'EasyFingerTouch', 'NimlyPRO', 'NimlyCode', 'NimlyTouch', 'NimlyIn'],`

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["Onesti Products AS", "Onesti Products AS"]
  "modelid": ["EasyFingerTouch", "EasyCodeTouch"],
  "vendor": "EasyAccess",
  "product": "Doorlock with keypad",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DOOR_LOCK_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x0b",
        "0x0101"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 11,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 11,
            "eval": "Item.val = Attr.val/2",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "config/lock",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "Item.val = Attr.val ==1",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lockstate",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "if (Attr.val == 0) { Item.val = 'not fully locked' } else if (Attr.val == 1) { Item.val = 'locked' } else if (Attr.val == 2) { Item.val = 'unlocked' } else { Item.val = 'undefined' }",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 11,
            "eval": "Item.val = Attr.val !=1",
            "fn": "zcl:attr"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 1,
          "max": 300
        },
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    }
  ]
}

```

If you use phoscon, pls take a look on the API direclty, using Phoscon/help/API Information/sensors. Phoscon is just a third app, we haven't impact on it.

About your last problem

> After a missing state change, the value is updated when you read it in deconz ? Cluster 0x0101 attribute 0x0000.
> I m seing this device have too state/open and state/lockstate (if they are not visible in HA can use Phoscon/help/API Information/sensors)
> If yes, it's a bind/report issue else the problem is in the DDF.

cortado, 2024-01-10:
alright! Got the state change to work after a while! Now it displays, with just minor delay - on lock/unlock :)  (not using Phoscon really - just for pairing devices)

So only question left is: should cluster 0xFEA2 be ignored, or is there something specific for this lock that could be explored there?

Smanar, 2024-01-10:
For me can be ignored.
I don't found a zigbee project that use it.

If all is working can you make a PR ? or want I make it ?

cortado, 2024-01-11:
The original EasyFIngerTouch PR can be updated with that it supports all these:  ['easyCodeTouch_v1', 'EasyCodeTouch', 'EasyFingerTouch', 'NimlyPRO', 'NimlyCode', 'NimlyTouch', 'NimlyIn'].

And should just be called "EasyAccess/Nimly locks" (they changed their brand)

Or if the right process is to make a new one, for supporting all, I suggest that - you are the expert here :) 

I have talked with Nimly support and they are happy to share a receipt I will give them, communication how to use HA + deCONZ with their locks :)

But will uses always have to manually install DDF, like I had to, or is this part of some distribution? (sorry for rookie level..)

Smanar, 2024-01-11:
>But will users always have to manually install DDF, like I had to, or is this part of some distribution? (sorry for rookie level..)

One time the DDF is valided/merged it will be in the officail version by defaut.

And yes as you haven't changed something, and use the same DDF for all device can just update the actual one.
You need to add As "manufacturername" than "modelid" so the result will be something like

```
  "manufacturername": ["Onesti Products AS", "Onesti Products AS", "NimlyPRO", "NimlyCode"]
  "modelid": ["EasyFingerTouch", "EasyCodeTouch", "EasyCodeTouch", "EasyCodeTouch"],
```

Just avoid the model "easyCodeTouch_v1" because this one is managed by the legacy code, so better to have an user with the device to be sure it don't break something.

cortado, 2024-01-15:
Great, so will you make the PR?

Smanar, 2024-01-16:
Sure > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7534

closed 2024-01-17 by manup